sony/flutter-embedded-linux #14: README and information on how to gather logs and errors

psstoyanov, 2021-03-10:
There is no information on how to gather logs or to trace errors.

To gather some further information about the DRM segfault I'm getting on my aarch64 devices, I've built the drm backend with an additional parameter so that lldb and gdb can provide more detailed feedback:

`cmake -DUSER_PROJECT_PATH=examples/flutter-drm-backend -DCMAKE_BUILD_TYPE=Debug ..`

What would be the preferred method of capturing/tracing errors with the embedder?

HidenoriMatsubayashi, 2021-03-17:
Thank you for your report. I'll update README.

HidenoriMatsubayashi, 2021-03-30:
I've added the debug mode. You can use `cmake -DUSER_PROJECT_PATH=examples/flutter-drm-backend -DCMAKE_BUILD_TYPE=Debug ..` to get gather logs and so on.

psstoyanov, 2021-03-30:
Checking [DEBUGGING.md](https://github.com/sony/flutter-embedded-linux/blob/master/DEBUGGING.md).

@HidenoriMatsubayashi , please correct me if I'm wrong. 
For the [Building with debug mode](https://github.com/sony/flutter-embedded-linux/blob/master/DEBUGGING.md#building-with-debug-mode) section - when building the embedder with `-DCMAKE_BUILD_TYPE=Debug` this would generate the debug symbols for the embedder itself, right? It won't have anything to do with the Flutter app build type. Probably it would be better to separate this section.

Note:
To further double-check my understanding, I've already tried and I was able to open the Dart debugger when the embedder has been built without specifying the debug flag. Debugging the Flutter app went as normal.
On the other hand, inserting break points inside the embedder itself or analysing an unexpected behaviour, can produce more information through `gdb` / `lldb` if the embedder is built with `DCMAKE_BUILD_TYPE`. 

For debugging the Flutter app and the [Command line](https://github.com/sony/flutter-embedded-linux/blob/master/DEBUGGING.md#command-line), maybe it will be good to add the note that the command has to be executed from the root folder of the Flutter project? ie:
```
$ cd <path_to_flutter_project>/
$ flutter attach --device-id=flutter-tester --debug-uri http://127.0.0.1:40409/k8IUol2dnPI=/
```

HidenoriMatsubayashi, 2021-03-31:
> when building the embedder with -DCMAKE_BUILD_TYPE=Debug this would generate the debug symbols for the embedder itself, right?

You are right.

> maybe it will be good to add the note that the command has to be executed from the root folder of the Flutter project? 

Exactly.

I'll update READMEs again.

psstoyanov, 2021-04-01:
Looks good! Thank you, @HidenoriMatsubayashi , @Mai-Matsuura !

Will close this ticket now.

closed 2021-04-01 by psstoyanov