sony/flutter-embedded-linux #15: DRM doesn't work the PinePhone

psstoyanov, 2021-03-10:
Using PinePhone running ManjaroARM with Plasma Mobile, the DRM backend doesn't work.
Device information:
- Allwinner A64
- kernel: `Linux plasma-mobile 5.11.4-1-MANJARO-ARM #1 SMP Sun Mar 7 19:22:26 UTC 2021 aarch64 GNU/Linux`
- mesa: `20.3.4` (the device is using the [Lima](https://gitlab.freedesktop.org/lima/web) driver)

Flutter version:
```
$ flutter --version
Flutter 2.1.0-11.0.pre.164 • channel unknown • unknown source
Framework • revision ede798e029 (2 days ago) • 2021-03-09 01:34:02 -0500
Engine • revision 2441c476a6
Tools • Dart 2.13.0 (build 2.13.0-116.0.dev)
```

LLDB trace when running the drm backend:
```
Process 29798 launched: '/home/kde/flutter-embedded-linux/build/flutter-drm-backend' (aarch64)
[ERROR][native_window_drm.cc(185)] Couldn't get resources
Process 29798 stopped
* thread #1, name = 'flutter-drm-bac', stop reason = signal SIGSEGV: invalid address (fault address: 0x20)
    frame #0: 0x0000aaaaaab2e570 flutter-drm-backend`flutter::NativeWindowDrm::FindConnector(this=0x0000aaaaaac75870, resources=0x0000000000000000) at native_window_drm.cc:215:34
   212 	}
   213 	
   214 	drmModeConnectorPtr NativeWindowDrm::FindConnector(drmModeResPtr resources) {
-> 215 	  for (int i = 0; i < resources->count_connectors; i++) {
    	                                 ^
   216 	    auto connector = drmModeGetConnector(drm_device_, resources->connectors[i]);
   217 	    // pick the first connected connector
   218 	    if (connector->connection == DRM_MODE_CONNECTED) {
```

I don't have a RPi4 to verify if it is the same issue as the one observed with https://github.com/sony/flutter-embedded-linux/issues/9

HidenoriMatsubayashi, 2021-03-11:
Thank you for your report.

Just now, we've fixed #9. So, could you see [here](https://github.com/sony/flutter-embedded-linux#run-with-drm-backend) and try again? You need to specify the drm device filename properly.

psstoyanov, 2021-03-11:
Thank you! Just had the chance to try the changes out.

Switching to tty3 and using `/dev/dri/card1` does create the Flutter window.
However, it crashes shortly after. Usually when I try to tap on the button.
The counter text is not visible too.

lldb log:
```
Process 6155 launched: '/home/kde/flutter-embedded-linux/build/flutter-drm-backend' (aarch64)
[WARNING][context_egl.h(95)] Failed to create EGL off-screen surface.(eglGetError: EGL_BAD_ALLOC)
[WARNING][linuxes_surface_gl_drm.h(50)] Off-Screen surface is invalid.
[ERROR][native_window_drm.cc(89)] TODO: implement here!!
[ERROR][flutter_linuxes_view.cc(60)] Failed to change surface size.
[ERROR:flutter/shell/platform/embedder/embedder_surface_gl.cc(102)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
[ERROR:flutter/shell/platform/embedder/embedder_surface_gl.cc(102)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter: Observatory listening on http://127.0.0.1:38703/rbJaBwNaqM8=/
[ERROR][native_window_drm.cc(311)] Unsupported cursor: 
[ERROR][native_window_drm.cc(176)] Could not move the mouse cursor: -13
[ERROR][native_window_drm.cc(130)] Failed to set cursor buffer. (-13)
[ERROR][native_window_drm.cc(176)] Could not move the mouse cursor: -13
[ERROR][native_window_drm.cc(130)] Failed to set cursor buffer. (-13)
[FATAL:flutter/fml/memory/weak_ptr.h(122)] Check failed: (checker_.checker).IsCreationThreadCurrent(). 
Process 6155 stopped
* thread #2, name = 'flutter-drm-bac', stop reason = signal SIGABRT
    frame #0: 0x0000fffff40c26c8 libc.so.6`raise + 216
libc.so.6`raise:
->  0xfffff40c26c8 <+216>: ldr    x5, [x5, #0xed8]
    0xfffff40c26cc <+220>: ldr    x1, [sp, #0x118]
    0xfffff40c26d0 <+224>: ldr    x0, [x5]
    0xfffff40c26d4 <+228>: eor    x0, x1, x0
```

HidenoriMatsubayashi, 2021-03-11:
Thank you so much. it seems like a bug of mouse cursor. Does it crash if you don't tap on the button?

psstoyanov, 2021-03-11:
From what I can see, it doesn't crash when I am not interacting with the app.

When I tap on the screen or do any keyboard or mouse keypress, the app will crash however.

I should add some animated widget in there :slightly_smiling_face: It is quite bland at the moment: 
<details>

<summary> PinePhone photo of the DRM embedder in action </summary>

[PinePhone_DRM_Mode](https://drive.google.com/file/d/1LC35500zWtLpazv-77XdCShBVhupyLb2/view?usp=sharing)

</details>

HidenoriMatsubayashi, 2021-03-12:
@psstoyanov Thank you.

@Mai-Matsuura Could you check and fix that?

```
[ERROR][native_window_drm.cc(311)] Unsupported cursor: 
[ERROR][native_window_drm.cc(176)] Could not move the mouse cursor: -13
[ERROR][native_window_drm.cc(130)] Failed to set cursor buffer. (-13)
[ERROR][native_window_drm.cc(176)] Could not move the mouse cursor: -13
[ERROR][native_window_drm.cc(130)] Failed to set cursor buffer. (-13)
[FATAL:flutter/fml/memory/weak_ptr.h(122)] Check failed: (checker_.checker).IsCreationThreadCurrent(). 
Process 6155 stopped
* thread #2, name = 'flutter-drm-bac', stop reason = signal SIGABRT
    frame #0: 0x0000fffff40c26c8 libc.so.6`raise + 216
libc.so.6`raise:
->  0xfffff40c26c8 <+216>: ldr    x5, [x5, #0xed8]
    0xfffff40c26cc <+220>: ldr    x1, [sp, #0x118]
    0xfffff40c26d0 <+224>: ldr    x0, [x5]
    0xfffff40c26d4 <+228>: eor    x0, x1, x0
```

HidenoriMatsubayashi, 2021-03-12:
@Mai-Matsuura I used the DRM backend on Ubuntu 20.04 desktop, but the mouse cursor won't be shown.

Mai-Matsuura, 2021-03-15:
@HidenoriMatsubayashi 

> I used the DRM backend on Ubuntu 20.04 desktop, but the mouse cursor won't be shown.

I've confirmed the mouse cursor is shown with the DRM backend on Ubuntu 20.04 desktop.
If you use a user who doesn't belong to the input group, the cursor will not be shown.
See: https://github.com/sony/flutter-embedded-linux/blob/master/README.md#note
If this is not the cause, please get a log to investigate the cause of the problem.

HidenoriMatsubayashi, 2021-03-15:
> If you use a user who doesn't belong to the input group, the cursor will not be shown.

Sorry, I've completely forgot that.

Mai-Matsuura, 2021-03-16:
@psstoyanov   
The log shows `Could not move the mouse cursor: -13`.
`-13` means "permission denied"(EACCES).
So, it seems that the cause of the error is not being able to become a DRM master.
I added the check of the DRM master, please retry using the latest flutter-embedded-linux code.
If the log shows `Couldn't become the DRM master.  ...`, please check your environment according to the log.

psstoyanov, 2021-03-16:
@Mai-Matsuura , will report tonight :+1:

psstoyanov, 2021-03-16:
@Mai-Matsuura , just checked on my PinePhone.
I can no longer observe the error `Could not move the mouse cursor: -13`.
I can't see a log message with `Couldn't become the DRM master.` neither.

When running the flutter app in release or profile mode:
- no issues, the cursor is visible, the text is visible, no crashes observed when moving the cursor, clicking or tapping on the button :+1: 

When running the flutter app in debug mode:
- I can't see the error for `Could not move the mouse cursor: -13` nor the log message for `Couldn't become the DRM master.`
- The app still crashes with the error for weak pointer
<details>
<summary> weak pointer error in debug mode </summary>

```
$ LD_LIBRARY_PATH=/home/kde FLUTTER_DRM_DEVICE="/dev/dri/card1" lldb ./flutter-drm-backend ./sample/build/linux/arm64/debug/bundle/
(lldb) target create "./flutter-drm-backend"
Current executable set to '/home/kde/flutter-embedded-linux/build/flutter-drm-backend' (aarch64).
(lldb) settings set -- target.run-args  "./sample/build/linux/arm64/debug/bundle/"
No entry for terminal type "xterm-termite";
using dumb terminal settings.
(lldb) run
Process 80258 launched: '/home/kde/flutter-embedded-linux/build/flutter-drm-backend' (aarch64)
[WARNING][context_egl.h(95)] Failed to create EGL off-screen surface.(eglGetError: EGL_BAD_ALLOC)
[WARNING][linuxes_surface_gl_drm.h(50)] Off-Screen surface is invalid.
[ERROR][native_window_drm.cc(97)] TODO: implement here!!
[ERROR][flutter_linuxes_view.cc(60)] Failed to change surface size.
[ERROR:flutter/shell/platform/embedder/embedder_surface_gl.cc(102)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
[ERROR:flutter/shell/platform/embedder/embedder_surface_gl.cc(102)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter: Observatory listening on http://127.0.0.1:44457/7vYHYiWX6W4=/
[FATAL:flutter/fml/memory/weak_ptr.h(122)] Check failed: (checker_.checker).IsCreationThreadCurrent(). 
Process 80258 stopped
* thread #2, name = 'flutter-drm-bac', stop reason = signal SIGABRT
    frame #0: 0x0000fffff41736c8 libc.so.6`raise + 216
libc.so.6`raise:
->  0xfffff41736c8 <+216>: ldr    x5, [x5, #0xed8]
    0xfffff41736cc <+220>: ldr    x1, [sp, #0x118]
    0xfffff41736d0 <+224>: ldr    x0, [x5]
    0xfffff41736d4 <+228>: eor    x0, x1, x0
```
</details>
Just tried the same setup running the app in debug mode on the PinebookPro and experienced the same crash - the crash should be in a new ticket from based on the latest results. I just happened to try it first on the PinePhone.

Note: On the PinebookPro after the app crashed, the tty session remained with black screen. Switching to tty4 and then back to tty3 returned the terminal session view.

@HidenoriMatsubayashi , @Mai-Matsuura , I believe this ticket can be closed as the drm mode works with the PinePhone now.

Mai-Matsuura, 2021-03-17:
@psstoyanov 
Thank you for your response. I'm glad your Pinephone has worked.
I confirmed debug mode on the DRM backend doesn't work in my environment (x64 ubuntu and RaspberryPi4) too.
I created #33 to investigate this issue.

I close this ticket, thank you for your support.

closed 2021-03-17 by Mai-Matsuura