sony/flutter-embedded-linux #21: [Question] Wayland client behavior and shell protocol

psstoyanov, 2021-03-11:
At the moment using the standalone wayland client appears to be made to work only within a Weston environment.
My attempts to use the embedder with other Wayland environments such as Plasma Wayland, [wlroots](https://github.com/swaywm/wlroots) based ones like [Wayfire](https://github.com/WayfireWM/wayfire) result in the following error:
```
[ERROR][linuxes_window_wayland.cc(627)] Invalid compositor and shell.
Failed to create view controller.
```
I haven't attempted to run the embedder within [Cage](https://github.com/Hjdskes/cage/wiki) or [Phosh](https://source.puri.sm/Librem5/phosh) but I don't expect different results.

From what I can see, this is because the defined shell protocol is specific to Weston's implementation: https://github.com/sony/flutter-embedded-linux/tree/master/src/wayland/protocol 

Would using a different shell implementation like [xdg](https://wayland-book.com/xdg-shell-basics.html) allow for more Wayland environments to be supported?
Or is the error given by the embedder a result from different check defined somewhere in [linuxes_window_wayland.h](https://github.com/sony/flutter-embedded-linux/blob/master/src/flutter/shell/platform/linux_embedded/window/linuxes_window_wayland.h)?

From the Wayland book, I can see the [section explaining](https://wayland-book.com/libwayland/wayland-scanner.html) on how to generate the C headers and glue code. Would experiments with other shell protocols be useful to the project? 

My understanding isn't as complete as I would like it to be so I don't know if I'm just using the Wayland client incorrectly in this case.

HidenoriMatsubayashi, 2021-03-11:
Thank you for your report. I'll test with other Wayland compositors.

psstoyanov, 2021-03-11:
Out of curiosity, is there a requirement for the wayland app to have references to a particular shell protocol?

I'm reading the following article from [Lup Yuen Lee](https://lupyuen.github.io/) (https://lupyuen.github.io/) which is packed with interesting information: https://lupyuen.github.io/pinetime-rust-mynewt/articles/wayland
Following the article and taking a look in the [LVGL repo](https://github.com/lupyuen/lvgl-wayland), I can't find any specific bindings to a particular shell protocol.

HidenoriMatsubayashi, 2021-03-12:
> Out of curiosity, is there a requirement for the wayland app to have references to a particular shell protocol?

Your understanding is correct. The implementation is not good. Actually, the Wayland app doesn't use shell protocols, but has referenced it.  Therefore, I'm going to refactor source files.
(cc @Mai-Matsuura)

HidenoriMatsubayashi, 2021-03-12:
@psstoyanov I'm using Ubuntu 20.04 desktop. Can I install Plasma Wayland or Wayfire? Do they work on X11?

psstoyanov, 2021-03-12:
@HidenoriMatsubayashi , I don't know what is the state of Plasma Wayland with Ubuntu 20.04 and aarch64. Hopefully it is in good condition?
 
Yes, you can start the wayland sessions from X11 environment.
If you go with Plasma Wayland - you can find the instructions on starting the session in their [docs](https://community.kde.org/Plasma/Wayland).

I would go with Sway or Wayfire as they are "easy" to handle if the distro repositories have broken or problematic builds with Plasma or Gnome. They are also significantly smaller (even building locally doesn't take unreasonable amount of time).

Both [Sway](https://github.com/swaywm/sway/wiki#configuration) (tiling) and [Wayfire](https://github.com/WayfireWM/wayfire/wiki/Configuration) (stacking) require a configuration file or you will be presented with an empty view. You can keep those minimal with pretty much defining just shortcut to launch your preferred/installed terminal.

You can also disable Xwayland on Sway/ Wayfire easily though I don't know if this will be useful to you.
Reference to the Xwayland flag for Wayfire [here](https://github.com/WayfireWM/wayfire/wiki/Configuration#corexwayland).

The example Wayfire configuration file provided in their repo: https://github.com/WayfireWM/wayfire/blob/master/wayfire.ini

This is [my wayfire configuration](https://gist.github.com/psstoyanov/bae71ba661d56776407a427abd3e0116) file - there is plenty in it as I use it as my main DE on the PinebookPro. 

[Cage](https://github.com/Hjdskes/cage) is tiny but I don't know if it will be flexible enough to use for this experiment - it is meant as a kios (though it could be a potential alternative to the weston-shell)

HidenoriMatsubayashi, 2021-03-12:
Thank you! I'll try that.

psstoyanov, 2021-03-12:
Forgot to mention something. Unlike Plasma Wayland, Sway and Wayfire are not full fledged desktop environments but compositors. The chances of breaking your test device setup are significantly smaller if go with one of them.

When I was experimenting with Gnome and Plasma side-by-side on the PinebookPro, I had to be extra careful about keeping them separate or some of the packages that handle the same functionality could cause hard to recover situations. Some references from the Manjaro's documentation on installing Desktop Environments [here](https://wiki.manjaro.org/index.php/Install_Desktop_Environments)

Wayfire has lived alongside Plasma on my PinebookPro without causing conflicts for more than a year now.

psstoyanov, 2021-03-16:
@HidenoriMatsubayashi , can see the wayland-client now works with Plasma Mobile, Plasma Wayland and Wayfire :fire: 

I can setup a few other images with different wayland sessions like Phosh, Gnome and so on if you would like some additional testing.
Should the ticket be closed?

HidenoriMatsubayashi, 2021-03-16:
Thank you for your help! Yes, I'm glad if you would do some additional testing. After that, I'll update README to add a tested Wayland compositors list.

psstoyanov, 2021-03-22:
@HidenoriMatsubayashi , had the chance to check Lomiri and Phosh tonight. Both run with the Wayland client without a problem.
Those 2 are in addition to Plasma Wayland, Plasma Mobile and Wayfire.

It's partially a guess but there will be some surprises with the display scaling setups on different DEs :smile: 
I'm sure Phosh runs with 200% scaling by default and it looks like Plasma Mobile does the same.

Switching to Lomiri, it mimicked 100% scaling on Phosh which would appear in similar size as the window visible on [this image](https://drive.google.com/file/d/1BlIvN6D-9xzjPSqN2x68bSxVKgbZNUAc/view?usp=sharing).

The desktop environment scaling on the device isn't connected to the embedder per-se but it could be good to know.

Tomorrow, I will load Gnome on one of the devices. If I can, I will try to take a look at SailfishOS or NemoMobile for the PinePhone

psstoyanov, 2021-03-23:
List of confirmed working environments with the Wayland client:
- [Gnome](https://www.gnome.org/) with Wayland backend enabled - distribution dependent
- [Phosh](https://source.puri.sm/Librem5/phosh) - Wayland shell for mobile devices like the [Librem5](https://puri.sm/products/librem-5/) and [PinePhone](https://www.pine64.org/pinephone/)
- [Plasma Wayland](https://community.kde.org/Plasma/Wayland) - a Wayland version of Plasma desktop environment
- [Plasma Mobile](https://www.plasma-mobile.org/) - open-source user interface for phones using KDE Frameworks
- [Lomiri](https://lomiri.com/) - Environment used mostly by [Ubuntu Touch](https://ubuntu-touch.io/) for mobile devices maintained by [UBports Community](https://www.ubports.com/community/about-the-community). Formely known as [Unity8](https://www.ubports.com/blog/ubport-blogs-news-1/post/lomiri-new-name-same-great-unity8-115). Desktop port of the environment is in the works.
- [Sway](https://swaywm.org/) - tiling Wayland compositor/ maintainers of [wlroots](https://github.com/swaywm/wlroots) project
- [Wayfire](https://wayfire.org/) - stacking Wayland compositor inspired by [Compiz](https://launchpad.net/compiz) based on [wlroots](https://github.com/swaywm/wlroots)
- [Cage](https://www.hjdskes.nl/projects/cage/) - Wayland kiosk. Runs a single maximized application. Supported Wayland protocols by Cage can be found [here](https://github.com/Hjdskes/cage/wiki#list-of-supported-protocols)

Known incompatible Wayland environment at the moment:
- [GlacierUX](https://wiki.merproject.org/wiki/Nemo/Glacier)

psstoyanov, 2021-03-23:
@HidenoriMatsubayashi , should I close the ticket?

The embedder is no longer bound to weston's shell protocols and the switch from `wl_shell` to `xdg_shell` has ensured compatibility with multiple environments.

HidenoriMatsubayashi, 2021-03-23:
Thank you so much! Your help is very helpful for us. I'll close this issue and update README later.

closed 2021-03-23 by HidenoriMatsubayashi

psstoyanov, 2021-03-25:
Update towards GlacierUX:
- It was using the deprecated `wl_shell` implementation - https://git.sailfishos.org/neochapay/lipstick/blob/nemo/src/compositor/lipstickcompositor.h#L23
- For the specific image that I've used, the developer is making similar jump to `xdg_shell`: https://github.com/neochapay/lipstick-1 
  - It's extremely likely that an updated image for the PinePhone with those changes to GlacierUX will function as well

HidenoriMatsubayashi, 2021-03-26:
Thank you for your information.

HidenoriMatsubayashi, 2021-05-11:
Thank you so much. I've updated README.