sony/flutter-embedded-linux #24: [README] Installing the embedder library and using different modes (debug/profile/release)

psstoyanov, 2021-03-12:
The documentation implies that `libflutter_engine.so` has to be installed here:
- https://github.com/sony/flutter-embedded-linux#install-flutter-core-embedder-library
and here:
- https://github.com/sony/flutter-embedded-linux/blob/master/BUILDING-ENGINE-EMBEDDER.md#5-install-embedder-library

While this is the preferred method, the libraries can be loaded without placing them in `/urs/lib`.
The following method can be used instead:
`LD_LIBRARY_PATH=<path_to_engine> ./flutter-client ./sample/build/linux/x64/debug/bundle`

This could be useful when:
- Switching quickly between debug / profile / release modes for the Flutter application without replacing `libflutter_engine.so`

HidenoriMatsubayashi, 2021-03-12:
Thanks. I'll update.

HidenoriMatsubayashi, 2021-03-15:
I've updated README.

HidenoriMatsubayashi, 2021-03-16:
The absolute path of `libflutter_engine.so` is specified in [the cmake file](https://github.com/sony/flutter-embedded-linux/blob/master/cmake/build.cmake#L119) to build. So, you need to install it to '/usr/lib' when you build. Is there any other good way?

psstoyanov, 2021-03-16:
@HidenoriMatsubayashi , I'm not sure about the build stage. I would prefer to spend some time checking the cmake documentation before making any suggestion. 

Resolved (partially) my issue when compiling the embedder library - I had a remnant from [flutter-git AUR](https://aur.archlinux.org/packages/flutter-git/) package on my x64 host system causing some conflicts with my main flutter install. Now the depot tools are working as they should at least towards my aarch64 devices and the bundled Dart VM is as expected.

Created the following setup on the PinebookPro (rk3399 system):
- Compiled the embedder with `libflutter_engine.so` (release)

- Removed `libflutter_engine.so` from `/usr/lib`

- Placed debug/profile/release engine libraries as:
 ```
➜  md5sum ~/flutter_engine/debug/libflutter_engine.so 
de613cd619f7c90c9ed0ebc3bad257ee  /home/pak0st/flutter_engine/debug/libflutter_engine.so
➜  md5sum ~/flutter_engine/profile/libflutter_engine.so 
3a5bc2a9bcdfb17a8b4e2626ba645643  /home/pak0st/flutter_engine/profile/libflutter_engine.so
➜  md5sum ~/flutter_engine/release/libflutter_engine.so 
f37b7067f8255a8217e73d5be1d3bf6e  /home/pak0st/flutter_engine/release/libflutter_engine.so
 ```

Running with proflile and release went well. Commands:
-  `LD_LIBRARY_PATH=/home/pak0st/flutter_engine/profile ./flutter-client ./sample/build/linux/arm64/profile/bundle`
[Profile mode](https://drive.google.com/file/d/1XefIk8DVcnVY9rD8azuFBEpThRxpDvzT/view?usp=sharing)
- `LD_LIBRARY_PATH=/home/pak0st/flutter_engine/release ./flutter-client ./sample/build/linux/arm64/release/bundle`
[Profile mode](https://drive.google.com/file/d/1hjk2uORh3TMbjaPmVOkNdVEoKOFy2Apz/view?usp=sharing)

The debug mode doesn't show any text for whatever reason:
- `LD_LIBRARY_PATH=/home/pak0st/flutter_engine/debug ./flutter-client ./sample/build/linux/arm64/debug/bundle`
[Debug mode](https://drive.google.com/file/d/1FHMhD3WpSNlbe1UfmycIKmw4Y_e6dKyT/view?usp=sharing)
No additional output when checking with gdb/lldb.

While not perfect, still a good progress I believe

HidenoriMatsubayashi, 2021-03-17:
Thank you for your help! Now, there seems to be an issue when working in debug mode. We are investigating to break down the issue whether it is a Flutter Engine issue or our embedder issue.

psstoyanov, 2021-03-24:
What's the difference between the custom `libflutter_engine.so` library and the one Flutter fetches on the arm64 device `libflutter_linux_gtk.so`?

HidenoriMatsubayashi, 2021-03-24:
`libflutter_linux_gtk.so` is for the official Flutter desktop for Linux using GTK. It includes `libflutter_engine.so`.

psstoyanov, 2021-03-24:
Aha, if I'm reading it correctly, this is the line which includes `libflutter_engine.so` : https://github.com/flutter/engine/blob/8c25ca1bfdba758137f102ea37e84291d5f0a74f/shell/platform/linux/BUILD.gn#L147

psstoyanov, 2021-03-26:
I was looking back through the commits of the `flutter-desktop-embedder` trying to find how the engine library was generated before they were pushed to the Flutter repository. I believe this the [one](https://github.com/google/flutter-desktop-embedding/tree/f3942495d1a1d37e9293a5614a3686a5ac17aeca/library). I wonder if something can be salvaged from there.

HidenoriMatsubayashi, 2021-03-28:
Thank you for your information. I'll check it from now.

HidenoriMatsubayashi, 2021-03-30:
I've updated READMEs.

psstoyanov, 2021-03-30:
Thank you, the READMEs are much clearer now :+1: 
I think this ticket is ready for closing.

closed 2021-03-30 by psstoyanov