sony/flutter-embedded-linux #30: [Question] Yocto support

jmarijan, 2021-03-16:
Do you have a plan to add a Yocto support for flutter-embedded-linux?

Yocto Layer for the Flutter already exists :
https://github.com/jwinarske/meta-flutter

It has support for some of the engines, such as flutter-pi.

HidenoriMatsubayashi, 2021-03-16:
Thank you for your question. Yes, I'm going to add the Yocto support for the example.

jmarijan, 2021-03-17:
Thank you for your response. Looking forward to it.

In meanwhile, I am trying to create a recipe for your repo on my own. I am using **dunfell**  version of Yocto.
There is a problem regarding **EGL/egl.h** library. While cross-compiling your repo on Yocto, following error is received:
:
`| tmp/work/aarch64-mx8mn-fslc-linux/flutter-embedded-linux/1.0-r0/git/src/flutter/shell/platform/linux_embedded/surface/context_egl.h:92:26: error: no matching function for call to 'eglCreateWindowSurface'
|     EGLSurface surface = eglCreateWindowSurface(
|                          ^~~~~~~~~~~~~~~~~~~~~~
|tmp/work/aarch64-mx8mn-fslc-linux/flutter-embedded-linux/1.0-r0/git/src/flutter/shell/platform/linux_embedded/surface/linuxes_surface_gl_drm.h:48:36: note: in instantiation of member function 'flutter::ContextEgl<gbm_device, gbm_surface, gbm_surface>::CreateOffscreenSurface' requested here
|     offscreen_surface_ = context_->CreateOffscreenSurface(window);
|                                    ^
| tmp/work/aarch64-mx8mn-fslc-linux/flutter-embedded-linux/1.0-r0/recipe-sysroot/usr/include/EGL/egl.h:155:31: note: candidate function not viable: cannot convert argument of incomplete type 'gbm_surface *' to 'EGLNativeWindowType' (aka 'wl_egl_window *') for 3rd argument
| EGLAPI EGLSurface EGLAPIENTRY eglCreateWindowSurface (EGLDisplay dpy, EGLConfig config, EGLNativeWindowType win, const EGLint *attrib_list);
`

In my opinion, your **EGL/egl.h** and **EGL/egl.h** library which Yocto fetches from source (https://archive.mesa3d.org/; version 20.0.2) are different. 
Can you please tell me which version of **EGL/egl.h** library is in use on your project?

psstoyanov, 2021-03-17:
@jmarijan , I believe EGLStream is already marked as pending for implementation in #1

HidenoriMatsubayashi, 2021-03-18:
@jmarijan
Did you add WL_EGL_PLATFORM define? NativeWindowType is different depends on the platforms. You need to specify it properly.

```
EGLSurface eglCreateWindowSurface(EGLDisplay display, 
    EGLConfig config, NativeWindowType native_window, EGLint const * attrib_list);
```

See: https://www.khronos.org/registry/EGL/api/EGL/eglplatform.h
```
#elif defined(WL_EGL_PLATFORM)

typedef struct wl_display     *EGLNativeDisplayType;
typedef struct wl_egl_pixmap  *EGLNativePixmapType;
typedef struct wl_egl_window  *EGLNativeWindowType;

#elif defined(__GBM__)

typedef struct gbm_device  *EGLNativeDisplayType;
typedef struct gbm_bo      *EGLNativePixmapType;
typedef void               *EGLNativeWindowType;

#elif defined(__ANDROID__) || defined(ANDROID)
```

TomohideMorimoto, 2021-03-22:
@HidenoriMatsubayashi   
Start working on this issue.

jwinarske, 2021-04-08:
I haven't smoke tested it on a target, but it builds.  https://github.com/jwinarske/meta-flutter/commit/04bcb9b83da1a5c6b7a2dda7244dad9218e9cbb4

@HidenoriMatsubayashi I understand you were asking around Toyota.  I own the Toyota Flutter Embedder.

HidenoriMatsubayashi, 2021-04-09:
@jwinarske 
Thank you for commenting and creating `flutter-embedded-linux_git.bb`. Actually, @TomohideMorimoto who is one of our member is creating a Yocto meta file for Flutter Engine and this embedder.

Yes, I talked a little with Toyota's members a few weeks ago. Are you one of them? Toyota's Flutter Embedder is focusing AGL, right? I'm interested in Flutter use in AGL and Toyota. Do you have any plans to open source? Especially, I'm interested in what to do with plugins such as video player and webview.

jwinarske, 2021-04-11:
> @jwinarske
> Thank you for commenting and creating `flutter-embedded-linux_git.bb`. Actually, @TomohideMorimoto who is one of our member is creating a Yocto meta file for Flutter Engine and this [...]

Open source plans are out of my hands.  If/when it does, it will happen through AGL (Yocto based).  The flutter-engine recipe won' change much beyond [this](https://github.com/jwinarske/meta-flutter/blob/dunfell/recipes-graphics/flutter-engine/flutter-engine_git.bb) one, if at all.

As far as plugins go I prefer manually loading a shared module, oppose to statically linking the plugin code.  If the plugin is in the filesystem, then it gets loaded, otherwise returns not implemented on the channel.  It keeps a clean demarcation between random plugin authors and application owners.  So having a plugin that can be built for GTK Linux or standalone plugin (shared module with known API including version) would meet the requirement.  Otherwise trying to cross compile a plugin using Flutter tools would require binary repository package changes not standard between organizations.  So alignment on plugins is a need.

HidenoriMatsubayashi, 2021-04-11:
I see. Thank you for sharing!

jwinarske, 2021-04-14:
Updated recipes [here](https://github.com/jwinarske/meta-flutter/tree/dunfell/recipes-graphics/sony). Each example app is a separate recipe.  Patches are specific to dunfell (libweston-8).  I renamed the bins so they can be installed at the same time.

There is currently an issue with XDG protocol hanging.  Seems you're missing a surface commit, which also indicates you're building with older Wayland Protocol and Weston.

HidenoriMatsubayashi, 2021-04-16:
@jwinarske 

> There is currently an issue with XDG protocol hanging. Seems you're missing a surface commit, which also indicates you're building with older Wayland Protocol and Weston.

Could you give me some more information? Do you mean that `wl_surface_commit` should be called every time?

HidenoriMatsubayashi, 2021-04-19:
@TomohideMorimoto Please check your build environment and fix properly.

topolittle, 2021-04-20:
I managed to build flutter-wayland-client under Yocto. Here what I have changed:
There was a missing build dependency. I added wayland-protocol to the DEPENDS variable.
I had to make some change to CmakeList.txt to use the Yocto's cross compiler instead of the native one. Here is the patch for it:
```
From b447ab6b15a9b906b0990eebbc7068ca410baa79 Mon Sep 17 00:00:00 2001
From: Stephane Gilbert <stephane.gilbert@exfo.com>
Date: Tue, 20 Apr 2021 13:08:51 -0400
Subject: [PATCH] Fix: Wrong clang compiler used in Yocto Setting
 CMAKE_C_COMPILER and CMAKE_CXX_COMPILER in CMakeList overrides the compiler
 binary Bitbake should use to cross-compile E.g.: clang++ vs
 aarch64-poky-linux-clang++

---
 CMakeLists.txt | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/CMakeLists.txt b/CMakeLists.txt
index 9008a04..1a7b44a 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -2,8 +2,8 @@ cmake_minimum_required(VERSION 3.10)
 project("flutter_linuxes" LANGUAGES CXX C)
 
 # Rules of compiler and cpp version.
-set(CMAKE_C_COMPILER clang)
-set(CMAKE_CXX_COMPILER clang++)
+#set(CMAKE_C_COMPILER clang)
+#set(CMAKE_CXX_COMPILER clang++)
 set(CMAKE_CXX_STANDARD 17)
 
 # Build options.
-- 
2.30.2
```

HidenoriMatsubayashi, 2021-04-22:
@topolittle 

> There was a missing build dependency. I added wayland-protocol to the DEPENDS variable.

I would like to confirm. wayland-protocol isn't explicitly added, but It exists in [meta/recipes-graphics/wayland](http://git.yoctoproject.org/cgit/cgit.cgi/poky/tree/meta/recipes-graphics/wayland). So, I think it's okay without it.

jwinarske, 2021-04-27:
A meta-layer is always in a separate source repo, otherwise you clone the same repo twice, and you end up with source in your meta layer.  Not a good pattern.

Why not just reference my [layer](https://github.com/jwinarske/meta-flutter)?  Not to mention it's going to get some updates.

HidenoriMatsubayashi, 2021-04-27:
Thank you for your advice. By the way, just wondering, is your activity on here in a position as Toyota or personal?

jwinarske, 2021-04-28:
Personal.

topolittle, 2021-05-07:
Following the commit ae9e9758d7, the above patch I posted to cross-compile in Yocto is no longer required.