sony/flutter-embedded-linux #33: Debug mode doesn't work on the DRM backend.

Mai-Matsuura, 2021-03-17:
Flutter applications in debug mode don't work on the DRM backend.  
The application crashes with the following log:
```
[FATAL:flutter/fml/memory/weak_ptr.h(122)] Check failed: (checker_.checker).IsCreationThreadCurrent(). 
```

Related: https://github.com/sony/flutter-embedded-linux/issues/15#issuecomment-800447168

Mai-Matsuura, 2021-03-22:
@psstoyanov
I fixed this issue. Flutter applications in debug mode work on the DRM backend using the latest flutter-embedded-linux.

psstoyanov, 2021-03-22:
Just tried it on the PinebookPro/ PinePhone with DRM and it worked well on both!

Thank you, @Mai-Matsuura , I believe this ticket can be closed :+1:

Mai-Matsuura, 2021-03-23:
I close this ticket. Thank you, @psstoyanov .

closed 2021-03-23 by Mai-Matsuura