sony/flutter-embedded-linux #34: Can't run Gallery app

psstoyanov, 2021-03-17:
Can't start the [Gallery](https://github.com/flutter/gallery) app using the embedder.
There are no log messages to help narrow down why that is:
```
$ ./flutter-client ./gallery/build/linux/x64/release/bundle     

```
Observed on my x64 system as well as the PinebookPro (rk3399) and PinePhone (Allwinner A64).

Tried with drm client and debug modes as well - still no feedback from the system as to why it doesn't want to start.

HidenoriMatsubayashi, 2021-03-17:
Thank you for your report. I'll check that later. (cc @Mai-Matsuura)

HidenoriMatsubayashi, 2021-03-18:
I have been able to reproduce this issue.

HidenoriMatsubayashi, 2021-03-18:
I need to do debugging, but I think this is a Flutter Engine issue.

psstoyanov, 2021-03-18:
Thanks, @HidenoriMatsubayashi , great find!

I've updated and recompiled the engine and Gallery app again and got slightly different results this time.

- `flutter-drm-backed`
  - Launched as expected with RX6800 (x64 system)
  - Launches as expected with PinebookPro (rk3399 system)
  - PinePhone spawned separate error for `settingsTextScaling` (should be separate issue with the system setup - will try to narrow it down later on and open a ticket) 

 <details>
 <summary> PinePhone error with Gallery in DRM mode </summary>
 
 ```
  $ FLUTTER_DRM_DEVICE="/dev/dri/card1" ./flutter-dr<ER_DRM_DEVICE="/dev/dri/card1" ./flutter-drm-backend                      ./gallery/build/linux/<1" ./flutter-drm-backend ./gallery/build/linux/arm64/                      release/bundle
[WARNING][context_egl.h(95)] Failed to create EGL off-screen surface.(eglGetError: EGL_BAD_ALLOC)
[WARNING][linuxes_surface_gl_drm.h(50)] Off-Screen surface is invalid.
[ERROR][native_window_drm.cc(97)] TODO: implement here!!
[ERROR][flutter_linuxes_view.cc(60)] Failed to change surface size.
[ERROR:flutter/shell/platform/embedder/embedder_surface_gl.cc(102)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
[ERROR:flutter/shell/platform/embedder/embedder_surface_gl.cc(102)] Could not create a resource context for async texture uploads. Expect degraded performance. Set a valid make_resource_current callback on FlutterOpenGLRendererConfig.
flutter: NoSuchMethodError: The getter 'settingsTextScaling' was called on null.
flutter: Receiver: null
flutter: Tried calling: settingsTextScaling
flutter: #0      _SettingsPageState.build (package:gallery/pages/settings.dart:146)
flutter: #1      StatefulElement.build (package:flutter/src/widgets/framework.dart:4689)
flutter: #2      ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:4572)
flutter: #3      StatefulElement.performRebuild (package:flutter/src/widgets/framework.dart:4744)
flutter: #4      Element.rebuild (package:flutter/src/widgets/framework.dart:4265)
flutter: #5      ComponentElement._firstBuild (package:flutter/src/widgets/framework.dart:4551)
flutter: #6      StatefulElement._firstBuild (package:flutter/src/widgets/framework.dart:4735)
flutter: #7      ComponentElement.mount (package:flutter/src/widgets/framework.dart:4546)
flutter: #8      Element.inflateWidget (package:flutter/src/widgets/framework.dart:3608)
flutter: #9      Element.updateChild (package:flutter/src/widgets/framework.dart:3363)
flutter: #10     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:4597)
flutter: #11     Element.rebuild (package:flutter/src/widgets/framework.dart:4265)
flutter: #12     ComponentElement._firstBuild (package:flutter/src/widgets/framework.dart:4551)
flutter: #13     ComponentElement.mount (package:flutter/src/widgets/framework.dart:4546)
flutter: #14     Element.inflateWidget (package:flutter/src/widgets/framework.dart:3608)
flutter: #15     Element.updateChild (package:flutter/src/widgets/framework.dart:3363)
flutter: #16     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:4597)
flutter: #17     StatefulElement.performRebuild (package:flutter/src/widgets/framework.dart:4744)
flutter: #18     Element.rebuild (package:flutter/src/widgets/framework.dart:4265)
flutter: #19     ComponentElement._firstBuild (package:flutter/src/widgets/framework.dart:4551)
flutter: #20     StatefulElement._firstBuild (package:flutter/src/widgets/framework.dart:4735)
flutter: #21     ComponentElement.mount (package:flutter/src/widgets/framework.dart:4546)
flutter: #22     Element.inflateWidget (package:flutter/src/widgets/framework.dart:3608)
flutter: #23     Element.updateChild (package:flutter/src/widgets/framework.dart:3363)
flutter: #24     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:4597)
flutter: #25     Element.rebuild (package:flutter/src/widgets/framework.dart:4265)
flutter: #26     ComponentElement._firstBuild (package:flutter/src/widgets/framework.dart:4551)
flutter: #27     ComponentElement.mount (package:flutter/src/widgets/framework.dart:4546)
flutter: #28     Element.inflateWidget (package:flutter/src/widgets/framework.dart:3608)
flutter: #29     Element.updateChild (package:flutter/src/widgets/framework.dart:3363)
flutter: #30     SingleChildRenderObjectElement.mount (package:flutter/src/widgets/framework.dart:6055)
flutter: #31     Element.inflateWidget (package:flutter/src/widgets/framework.dart:3608)
flutter: #32     Element.updateChild (package:flutter/src/widgets/framework.dart:3363)
flutter: #33     ComponentElement.performRebuild (package:flutter/src/widgets/framework.dart:4597)
flutter: #34     StatefulElement.performRebuild (package:flutter/src/widgets/framework.dart:4744)
flutter: #35     Element.rebuild (package:flutter/src/widgets/framework.dart:4265)
flutter: #36     ComponentElement._firstBuild (package:flutter/src/widgets/framework.dart:4551)
flutter: #37     StatefulElement._firstBuild (package:flutter/src/widgets/framework.dart:4735)
flutter: #38     ComponentElement.mount (package:flutter/src/widgets/framework.dart:4546)
flutter: #39     Element.inflateWidget (package:flutter/src/widgets/framework.dart:3608)
flutter: #40     Element.updateChild (package:flutter/src/widgets/framework.dart:3363)
[... cut ...]
flutter: #98     RenderObject.layout (package:flutter/src/rendering/object.dart:1783)
flutter: #99     RenderProxyBoxMixin.performLayout (package:flutter/src/rendering/proxy_box.dart:117)
```
</details>

- `./flutter-client` 
  - With all 3 systems it doesn't start.
  - However with your find of resizing the Weston window, I've switched from and back to the same tty session and the app was present. 

Possibly this has something to do with the Gallery app being one of the examples for adaptive UI (mobile/ desktop)? It's just a guess but maybe #6 is connected

HidenoriMatsubayashi, 2021-03-19:
I've confirmed that it works fine when I use the DRM backend on my desktop.

> Possibly this has something to do with the Gallery app being one of the examples for adaptive UI (mobile/ desktop)?

Yes, I think so.

HidenoriMatsubayashi, 2021-03-26:
I've tried with the x11 backend which I've been developing now, but it works fine..

psstoyanov, 2021-03-26:
Thanks, that's good to know.

I will set aside a bit of time this Sunday to try to gather some further information. Possibly go through the other example app that deals with mobile/ desktop optimisation to see if it's limited just to the Gallery app.

HidenoriMatsubayashi, 2021-05-13:
@TomohideMorimoto

jwinarske, 2021-05-19:
I can run the Gallery app (master) AOT on a NVIDIA Xavier (demo-image-egl - master) using flutter-drm-eglstream-backend.  LC_ALL needs to be properly set or you get a gray screen with a small flutter logo on top of the screen.  Mouse drags don't work, only click and scroll.

root@jetson-xavier-nx-devkit:~# LC_ALL=en_GB flutter-drm-eglstream-backend -b /usr/share/flutter-gallery/sony
[WARNING][linuxes_window_drm.h(112)] FLUTTER_DRM_DEVICE is not set, use /dev/dri/card0
[INFO][native_window_drm.cc(67)] resolution: 1680x1440
[INFO][linuxes_window_drm.h(140)] Display output resolution: 1680x1440
[ERROR][native_window_drm.cc(43)] Couldn't move the mouse cursor: -14
[ERROR][native_window_drm.cc(43)] Couldn't move the mouse cursor: -14
[ERROR][native_window_drm.cc(43)] Couldn't move the mouse cursor: -14
[ERROR][native_window_drm.cc(43)] Couldn't move the mouse cursor: -14
[ERROR][native_window_drm.cc(43)] Couldn't move the mouse cursor: -14
[ERROR][native_window_drm.cc(43)] Couldn't move the mouse cursor: -14
[ERROR][native_window_drm.cc(43)] Couldn't move the mouse cursor: -14
[ERROR][native_window_drm.cc(43)] Couldn't move the mouse cursor: -14
[ERROR][native_window_drm.cc(43)] Couldn't move the mouse cursor: -14
[ERROR][native_window_drm.cc(43)] Couldn't move the mouse cursor: -14
[ERROR][native_window_drm.cc(43)] Couldn't move the mouse cursor: -14

jcviau, 2021-05-20:
I have been able to run the gallery application on an i.MX 8M Mini with wayland. I had to remove one line in main.dart, otherwise the application isnt displayed. The line is: `restorationScopeId: 'rootGallery'`

For us, the **main problem** is related to the touch gesture. I asked someone else to check if he has the same issues and I will do additional test but so far:
- Scroll/drag continue randomly once I remove my finger (lists and 2D transformation when moving the object)
- Multi-touch isnt working for rotation, pin zoom/pan (tested in 2D transformation page). 

**Other issues** I saw:
**Mouse-only issue**
- A few operation (x axis motion, 2d transformation) raised errors in the console: Unsupported cursor: Forbidden
- The snackbar control raise an exception when clicking the action text on the snackbar
**Touch & Mouse issue**
- Bottom App Bar is displayed correctly but keep showing errors in the console: _some draw get error and skipped during validation_ 
- Any click in the Shrine page raise an exception.
In Debug: _Cannot hit test a render box that has never been laid out._
In Release: _The method 'createTicker' was called on null._
	
Note: I will update this post shortly after my colleague try the gallery application on his i.MX 8M Mini.

jwinarske, 2021-05-20:
> I have been able to run the gallery application on an i.MX 8M Mini with wayland. I had to remove one line in main.dart, otherwise the application isnt displayed. The line is: `restorationScopeId: 'rootGallery'`
> 

I found if 'LC_ALL' (locale) is not set properly it just displays a gray screen.  I didn't have to edit any Dart code.  This could also be impacting the Shrine app etc.

> For us, the **main problem** is related to the touch gesture. I asked someone else to check if he has the same issues and I will do additional test but so far:
>
>     * Scroll/drag continue randomly [...]

Which embedder are you running, and what flavor/version of compositor?

jcviau, 2021-05-20:
> Which embedder are you running, and what flavor/version of compositor?
Embedder is flutter-wayland with weston as the compositor

jcviau, 2021-05-20:
> I found if 'LC_ALL' (locale) is not set properly it just displays a gray screen. I didn't have to edit any Dart code. This could also be impacting the Shrine app etc.

I tried with LC_ALL set but I still had to remove the same line in main.dart to make the application appear. It's not even gray, it's not visible at all when this line is present.

jwinarske, 2021-05-20:
Interesting.

@HidenoriMatsubayashi san.  There should be a bit more debug output around platform channels, and callbacks in debug builds.  Masking these is not a good idea, as it can directly affecting program behavior.  Some channel callbacks and engine callbacks are required for proper operation of the gallery app.  Also it might be good to have bug template that requests weston and wayland versions from reporter.

@jcviau Non-modified weston 8/9 doesn't work for me in any scenario, regardless the app.  It should work unaltered on desktop embedder, as well as any other available embedder.  So if alteration to make app run specific to to one of these embedder it should be reported as a unique bug.  It very well could be related to non-implemented platform channels / engine callbacks.

jwinarske, 2021-05-20:
> Is there a way to diagnostic those issues (application not shown) ? Would
> allow us to create a new issue with more details. Per example, is there
> some logs we can collect and attach to the issue ?

Yes I agree that would make sense

psstoyanov, 2021-05-21:
> We were able to run the gallery application on an i.MX 8M Mini with wayland. We had to remove one line in main.dart, otherwise the application isnt displayed. The line is: `restorationScopeId: 'rootGallery'`

This does appear to be the cause of the problem with the Wayland embedder. I haven't used [RestorationManager](https://api.flutter.dev/flutter/services/RestorationManager-class.html) thus far unfortunately.

This ticket is getting a bit overloaded. Perhaps it will be better to split it into several different ones.

HidenoriMatsubayashi, 2021-05-21:
@jcviau @psstoyanov @jwinarske 

Thanks a lot. We were also investigating this issue. As @jcviau mentioned, we found that problems occur when using [`restorationScopeId` property](https://api.flutter.dev/flutter/widgets/Navigator/restorationScopeId.html). We have confirmed that the same problem occurs with [a simpler sample](https://dev.to/pedromassango/what-is-state-restoration-and-how-to-use-it-in-flutter-5blm).

HidenoriMatsubayashi, 2021-05-21:
@jcviau 

> For us, the main problem is related to the touch gesture.

Thank you for reporting a bug. I created a new issue for this bug. https://github.com/sony/flutter-embedded-linux/issues/127

HidenoriMatsubayashi, 2021-05-21:
We found out why the Gallery app doesn't work properly. Using `wl_display_dispatch` is the reason for not work Gallery app. The events from Engine while waiting for Wayland events using [`wl_display_dispatch`](https://github.com/sony/flutter-embedded-linux/blob/master/src/flutter/shell/platform/linux_embedded/window/linuxes_window_wayland.cc#L706) aren't handled...

closed 2021-05-24 by HidenoriMatsubayashi

bitinerant, 2022-08-06:
> Note: You can also hard code the locale in the application by changing locale:null (in main.dart) to something else. e.g. locale: const Locale('en', ''). That's what I did initially instead of setting LC_ALL.

I've been trying to run the v2.9.2 release for hours. I tried building from source, the [pre-built Linux binaries](https://github.com/flutter/gallery/releases/download/v2.9.2/flutter_gallery_Linux.tar.gz), and the Snap (`sudo snap install flutter-gallery`). All of these gave me the big empty grey window until I made the above code change. I'm running Ubuntu 22.04. I hope this doesn't discourage potential new Flutter users.

PR #131 (2021-05-24): Fixed bug blocking events from flutter engine for Wayland
Fixed #34, #132 

I'm going to refactor the main loop code...
commit 1cab5426 (2021-05-24): Fixed bug blocking events from flutter engine for Wayland (#131)

Fixed #34, #132