sony/flutter-embedded-linux #36: Ozone abstraction

oillipheist, 2021-03-18:
Hi, 
I was wondering if you might be interested in using Ozone to abstract away the back ends for Wayland, DRM, etc. 

That would also bring along support for X11, Cast and any future backends as well.

There is no dependency on any other toolkit like GTK or even a dependency on Chromium.

It would bring along support for running headless, using hardware overlays and zero-copy texture uploads for things like low power hardware accelerated video on embedded and desktop platforms.

HidenoriMatsubayashi, 2021-03-19:
Thank you for sharing with me. I've heard Ozone, but I don't know about it well. It seems a part of the source code in Chromium, right? Is there a stand-alone project?

oillipheist, 2021-03-19:
It’s a stand alone project,  it’s on the final stretch towards being enabled by default for Chromium. 

Chromium will depend on Ozone but Ozone does not depend on Chromium. 
Flutter and other GUI frameworks could use Ozone too without depending on Chromium.

In Chromiums case, once it’s enabled by default there will be opportunity to remove GtkUi (which is the legacy way in which Chromium accessed the native file picker) and replace with a service based approach (XDG-Desktop portals)). 

This will allow Chromium to remove the hard dependency on Glib and GTK. 

It would also present an opportunity to distribute Ozone Chrome, CEF and Flutter applications in fully sandboxed containers(e.g Flatpak, Snaps).

XDG-Desktop-Portals would provide standards for punching holes safely through the sandbox in order to get the native file picker, app picker, access the microphone, webcam, location data etc

Here is a document discussing a migration of Chromium from GtkUi to a service based approach with ozone - https://docs.google.com/document/u/0/d/1sdOlPa9_dS3hOBplpKU2be72ZAHy1zZJrU46zcU49Xs/mobilebasic#

And here is some more information on xdg-portals and what kind of stuff you can do with them - https://flatpak.github.io/xdg-desktop-portal/portal-docs.html

HidenoriMatsubayashi, 2021-03-19:
Sounds good! It seems worth considering. Is there a simple example code using Ozone?

oillipheist, 2021-03-19:
Yeah, the neat thing about going with a service based approach for things like file picker is that you will be able to create a backend in flutter for the GUI portals. If a user for e.g installs a third party application and makes a request for the native file picker then your flutter file picker is what will get launched regardless of whether that application is written using GTK, Qt, Electron, EFL etc.

On the desktop its useful for the user to be able to restrict what access an application has outside of its sandbox(e.g webcam, microphone etc) but it could also be useful for e.g automotive use cases where you might want to set system wide policies whilst also opening the platform for third parties.(e.g preventing webcam access when car is moving, or preventing an application from inhibiting the session when moving, running in foreground or background etc)).

XDG-Desktop-Portals would make all of that possible and could be adopted regardless of whether you decide to use Ozone or not. 

With regard to Ozone, the two main consumers of it right now are Cef and Chromium.
The interfaces which you would need to support are discussed over at https://chromium.googlesource.com/chromium/src.git/+/master/docs/ozone_overview.md .

Igalia have done allot of work on it over the last couple of years to get it robust enough to be switched on by default for Chrome, some of this work has involved fixing bugs with Ozone itself and some of it has involved trying to extend the Wayland protocol to support new use cases.

On top of the Cast, X11, Wayland and DRM backends there is also a Scenic backend for Fuchsia.

Perhaps @msisov and @tonikitoo could give this a once over to make sure that this would be a worthwhile approach.

Thank you for working on the Wayland embedder by the way. 👍🏻😃

msisov, 2021-03-22:
Ozone hasn't really been used without Chromium yet. I'm not sure it was even considered to be used without Chromium. Though, you're right. There are very few dependencies and some of them are base and ui/base components that provide some Linux fundamental functions wrapped in a nice C++ way that Chromium also uses.

CEF uses Chromium as a whole. Basically, Chromium is a library for that project. Thus, whatever Chromium holds (Ozone), CEF uses.

oillipheist, 2021-03-22:
Ozone-demo would be an example of Ozone used without Aura right?
I know that display configuration was an issue in the past for quickly getting stuff bootstrapped and displaying on screen without ash.

That seems to have been [solved](https://medium.com/joone/implement-a-single-interface-for-ozone-gbm-clients-to-configure-displays-d9e835b33617) though.

I'm just going to link to multi-window spec [here](https://docs.google.com/document/u/0/d/11_4wntz_9IJTQOo_Qhp7QF4RfpIMTfVygtOTxQ4OGHY/mobilebasic) as its somewhat relevant towards reasoning how the embedder should look whether its built using Ozone or not.

Its a bit fiddly to search for atm so its nice to have it referenced in an issue over here.

I feel that for knowledge on integrating with windowing systems across platforms and the edge cases that one might come across, Ozone is definitely worth keeping an eye on. Maybe even have shared DRM/Wayland and potentially Cocoa and Win32 backends across both Flutter and Chromium.
 
Speaking of which, feel free to close this issue if you feel that it is out of scope at this moment in time or in general and thank you again for working on the Wayland/DRM embedder.

oillipheist, 2021-03-26:
Just going to close this issue to reduce noise on the tracker, thanks for all of your hard work again.

closed 2021-03-26 by oillipheist

HidenoriMatsubayashi, 2021-03-26:
Thank you for your information!

jwinarske, 2021-04-11:
> Ozone-demo would be an example of Ozone used without Aura right?
> I know that display configuration was an issue in the past for quickly getting stuff bootstrapped and displaying on screen without [...]

Ozone plays the role of what GTK does today for Flutter Desktop Linux.  Ozone has been around for a long time with little to no traction.  I consider GTK too big for embedded use cases, as I CEF.  Content_shell has a smaller runtime footprint, starts faster, and JS<->Native calls are supported.

HidenoriMatsubayashi, 2021-04-11:
> I consider GTK too big for embedded use cases

I agree with you.

> Content_shell has a smaller runtime footprint, starts faster, and JS<->Native calls are supported.

I don't know about Content_shell well, but it looks like one of Chromium modules.

jwinarske, 2021-04-11:
> > Content_shell has a smaller runtime footprint, starts faster, and JS<->Native calls are supported.
> 
> I don't know about Content_shell well, but it looks like one of Chromium modules.

It's an optional build target when building chrome, targeting testing.  When I was at Inrix/OpenCar (HTML5 auto UI that's what I used.  The RAM runtime footprint was down to around 115MB with ~72MB of minified Java Script.  Native-JS was the key to usability.

HidenoriMatsubayashi, 2021-04-11:
That makes sense but, JS isn't needed in this case (Flutter UI), right?

jwinarske, 2021-04-12:
Only in case where you're implementing an HTML5 UI that needs to talk to the back end.