sony/flutter-embedded-linux #41: Support Platform views

HidenoriMatsubayashi, 2021-03-22:
We need to add support Platform views support to embed native views like WebView in a Flutter app

jwinarske, 2021-04-11:
[Hybrid Composition](https://github.com/flutter/flutter/wiki/Hybrid-Composition).  Please read the Performance section.

The problem with the embedded space is that one size doesn't fit all.  So having a generic Platform view solution may not meet the general communities needs.  Please keep this in mind when looking at potential solutions.

CodeDoctorDE, 2023-05-11:
What's the status of this issue?

HidenoriMatsubayashi, 2023-08-22:
Sorry for no updates.

Platform views implementation (embedder side) has already been merged into this repo. On Dart side, I'll merge https://github.com/sony/flutter-elinux/pull/206 (this new package for platform views) when I release next flutter-elinux version (3.13.1).

Thanks. Closing.

closed 2023-08-22 by HidenoriMatsubayashi