sony/flutter-embedded-linux #54: libflutter_engine.so install location confusion

Charlie9830, 2021-04-01:
Just finished running through the instructions to get an App built for an RPi4. Although I now have the app running (via drm backend) I did have some confusion with where to put the built libflutter_engine.so file.

I could only get it to run correctly if I copied libflutter_engine.so to both the cmake build directory as well as /usr/lib on the target device. I'm not sure if this is due to a bug or intended behavior. If it is intended behavior then I feel like the documentation needs some adjustment to reflect that. 
Perhaps Step 5 of ```BUILDING-ENGINE-EMBEDDER.md``` could be amended to also include installing libflutter_engine.so to /usr/lib on the target device.

It would also be worth amending step 3) of the ```Install Flutter Engine library``` section of the readme for those who download the engine from google infra.

HidenoriMatsubayashi, 2021-04-01:
> I could only get it to run correctly if I copied libflutter_engine.so to both the cmake build directory as well as /usr/lib on the target device.

You need to install `libflutter_engine.so` to the path which is added ld-path. You can also use `LD_LIBRARY_PATH`.

e.g.
```Shell:
LD_LIBRARY_PATH=./flutter_engine/ FLUTTER_DRM_DEVICE="/dev/dri/card1" ./flutter-drm-backend ./sample/build/linux/x64/profile/bundle
```

See also: https://github.com/sony/flutter-embedded-linux#supplement

Charlie9830, 2021-04-03:
Thanks Hidenori,

I had skipped over the supplement step as I wasn't planning at this stage to have to quickly switch between engine modes.

HidenoriMatsubayashi, 2021-04-16:
Closing this issue. If you have any question, please comment and I can re-open.

closed 2021-04-16 by HidenoriMatsubayashi