sony/flutter-embedded-linux #55: BUILDING-ENGINE-EMBEDDER.md missing ninja-build install step

Charlie9830, 2021-04-01:
Only a small issue but the ```BUILDING-ENGINE-EMBEDDER.md``` is missing a step to install ninja-build.
This caused a minor bit of confusion for myself as I'm new to Linux and C/C++ building.

HidenoriMatsubayashi, 2021-04-01:
Did you do https://github.com/sony/flutter-embedded-linux/blob/master/BUILDING-ENGINE-EMBEDDER.md#1-install-build-tools ? I think `ninja` is included `depot_tools`.

psstoyanov, 2021-04-01:
@Charlie9830 , the `depot_tools` include automated scripts to install all required dependencies. I haven't checked the automated scripts in detail as I'm not running a compatible distribution for them.

Reference in the documentation: https://commondatastorage.googleapis.com/chrome-infra-docs/flat/depot_tools/docs/html/depot_tools_tutorial.html#_example_walkthrough
Location of the install scripts: `cd <depot_tools>/src/build`

Notes:
- Those automated install scripts **are** distribution dependent and to some extent version dependent. Ideally, you will run them on Debian based systems (LTS versions of Ubuntu or Debian). If you are using something else - for example Fedora, Manjaro or Gentoo, you will need to install the required packages manually as most likely they will be different. In those cases it might be better to find organically which package is not present on your system.
- To avoid such problems with the required packages on the x64 host machine, you could use chroot environment (a container) to build the Flutter embedder library though managing containers can be intimidating for newcomers and poses it's own set of challenges.
- Different distribution releases, can include different set of libraries by default.

I hope the above information will be helpful to you.

HidenoriMatsubayashi, 2021-05-13:
I'll this issue. If you have any problem, I'll reopen it.

closed 2021-05-13 by HidenoriMatsubayashi