sony/flutter-embedded-linux #56: Host Platform Target Plaform Documentation

Charlie9830, 2021-04-01:
I've just finished successfully building for an RPi4 with the DRM Backend. Thanks for all your hard work guys.
I did have a small head scratching hiccup during the process though. I had to perform Step 3 ```Build for DRM Backend``` on the target device (RPi4). With a libflutter_engine.so targeting arm64, this step of the build would fail on my X64 machine. 
If this is expected behavior then I think it would be worth adding a note to Step 3 that the following steps must be built on the target platform (Similar to the note about compiling the Dart/Flutter code)

HidenoriMatsubayashi, 2021-04-01:
Thank you for your comments.

> With a libflutter_engine.so targeting arm64, this step of the build would fail on my X64 machine.

Did you try to cross-build for arm64 on x64 hosts, right?

I'll add add cross-build support and update README.

Charlie9830, 2021-04-03:
Hi Hidenori,

I think where my confusion lies is around step 3. Does step 3 have to be performed on the target platform? I understand that the Flutter project build ```flutter build linux``` does not allow for cross-building (currently). Is this also the case for Step 3?

HidenoriMatsubayashi, 2021-04-04:
Yes, you are right. THE current README is not good, so I'll update it.

HidenoriMatsubayashi, 2021-05-08:
We've updated README. https://github.com/sony/flutter-embedded-linux/pull/101

closed 2021-05-08 by HidenoriMatsubayashi