sony/flutter-embedded-linux #59: Create Flutter tools for embedded Linux

HidenoriMatsubayashi, 2021-04-04:
Currently, we need to build Flutter Engine to use this embedder ourselves. We also need to use a specific Flutter SDK to build Flutter apps. These tasks are bothered, so we will create a tool to support such as `flutter build embedded-linux` and `flutter run -d embedded-linux`, etc.

HidenoriMatsubayashi, 2021-05-17:
It seems that the official custom devices support in Flutter SDK is useful for us.

- https://github.com/flutter/flutter/issues/65067
- https://github.com/flutter/flutter/pull/78113
- https://github.com/flutter/flutter/issues/79406

HidenoriMatsubayashi, 2021-05-17:
Custom devices in Flutter SDK works fine.

```Shell
$ flutter config --enable-custom-devices
$ touch ~/.flutter_custom_devices.json
```

`~/.flutter_custom_devices.json` which is an example I've tried with my localhost.
```Json
{
  "custom-devices": [
    {
      "id": "embedded-linux",
      "label": "Embedded Linux Device",
      "sdkNameAndVersion": "Target Embedded Linux Device",
      "disabled": false,
      "ping": [
        "ping",
        "-w",
        "500",
        "-c",
        "1",
        "localhost"
      ],
      "pingSuccessRegex": "ttl=",
      "postBuild": null,
      "install": [
        "scp",
        "-r",
        "${localPath}",
        "user@localhost:/tmp/${appName}"
      ],
      "uninstall": [
        "ssh",
        "user@localhost",
        "rm -rf \"/tmp/${appName}\""
      ],
      "runDebug": [
        "ssh",
        "user@localhost",
        "<path>/flutter-client -b /tmp/${appName}"
      ],
      "forwardPort": [
        "ssh",
        "-o",
        "ExitOnForwardFailure=yes",
        "-L",
        "127.0.0.1:${hostPort}:127.0.0.1:${devicePort}",
        "user@localhost"
      ],
      "forwardPortSuccessRegex": "Linux",
      "screenshot": [
        "ssh",
        "user@localhost",
        "fbgrab /tmp/screenshot.png && cat /tmp/screenshot.png | base64 | tr -d ' \\n\\t'"
      ]
    }
  ]
}
```

```Shell
$ flutter devices
3 connected devices:

Linux (desktop)                • linux          • linux-x64      • Ubuntu 20.04.2 LTS 5.4.0-66-generic
Chrome (web)                   • chrome         • web-javascript • Google Chrome 89.0.4389.82
Embedded Linux Device (mobile) • embedded-linux • linux-arm64    • Target Embedded Linux Device

$ flutter run -d embedded-linux --verbose
```

HidenoriMatsubayashi, 2021-05-18:
At least now, I think [flutter-tizen](https://github.com/flutter-tizen/flutter-tizen)'s approach is better than using custom devices in Flutter SDK. Because custom devices in Flutter SDK don't support building a custom embedder. However, if we develop Android apps, iOS apps, Linux desktop apps, and Embedded Linux at the same time, we need to use two tools...

HidenoriMatsubayashi, 2021-07-16:
I published [flutter-elinux](https://github.com/sony/flutter-elinux).

closed 2021-07-16 by HidenoriMatsubayashi

doesnotexist, 2022-09-27:
Does that mean that if this PR https://github.com/flutter/flutter/pull/90958 or something resembling this PR was implemented in the upstream flutter tool, then the custom tool / flutter tool fork that is implemented in the flutter-elinux repository could be replaced by a smaller/simpler flutter tool plugin to allow the official flutter tool to build and run against the custom elinux embedder?

christopherfujino, 2022-09-29:
I suspect you'd still need to maintain your fork.

doesnotexist, 2022-10-05:
Ok, that makes sense.  Since commands like `flutter-elinux build` support flags such as --target-backend-type 
and that download the artifacts are retrieved from https://github.com/sony/flutter-embedded-linux/releases or an alternative specified by an environment variable ELINUX_ENGINE_BASE_URL.

Though it does appear that providing specifying artifact urls was considered for custom-devices https://github.com/flutter/flutter/issues/79406#issuecomment-843235080

I wonder if deprecating the fork is possible if custom-devices supported specifying artifact urls and something like the keys in the aformenetioned PR, namely `configureNativeProject` and `buildNativeProject`.  Would those keys or something very similar supplant the custom flags like --target-backend-type?

I'm not sure what other issues are solved by the fork. Perhaps VSCode Extension mentioned above?
> "Also, Extension for VSCode is an issue."