sony/flutter-embedded-linux #65: Display resolution change is not reflected with the EGLStream backend.

Mai-Matsuura, 2021-04-12:
If the resolution of the display changes while displaying the screen with the EGLStream backend, the screen display is not updated.

HidenoriMatsubayashi, 2021-04-12:
This problem will happen both GBM and EGLStream. We need to implement correctly.

Mai-Matsuura, 2021-04-21:
I will start deal with this issue.

HidenoriMatsubayashi, 2021-04-21:
Can you fix GBM first? I think it is easier than EGLStream.

Mai-Matsuura, 2021-04-21:
Yes, I will fix GBM first.

HidenoriMatsubayashi, 2021-04-21:
We need to detect to change a DRM connector first. After that, if the display resolution is changed and the new resolution is smaller than the current frame buffer size, we need to resize. If the new resolution is lager than the current frame buffer size, we don't need to do that, which means the frame buffer is the same size.

HidenoriMatsubayashi, 2021-05-17:
TODO: add resize support for DRM-EGLStream