sony/flutter-embedded-linux #68: vsync

jwinarske, 2021-04-13:
vsync is missing

HidenoriMatsubayashi, 2021-04-14:
Thank you. Actually, I don't understand what to use FlutterEngineOnVsync for. Could you tell me the purpose of it?

guangoon, 2021-04-14:
VSync can be understood as a timed interrupt. The system will send a VSync signal every time it can proceed, and the CPU/GPU will immediately process the rendering after receiving the signal. Therefore, in Flutter, when rendering is needed, we need to monitor the VSync signal, and when a VSync signal is generated, we can start GPU rendering. and if we don't set VsyncCallback in embedder, engine will set a default  VsyncCallback which is 60 fps.

HidenoriMatsubayashi, 2021-04-14:
Thank you. That's quite understandable. What should this embedder do is to prepare the API so that the user program or plugins can receive Vsync?

guangoon, 2021-04-14:
FlutterProjectArgs args = {};
  **args.vsync_callback =**
  you should implement this callback, 
  When this callback is triggered, you need request vsync signal, 
  when receive vsync signal callback, you need call FlutterEngineOnVsync.

HidenoriMatsubayashi, 2021-04-14:
Thanks. I'll implement vsync.

HidenoriMatsubayashi, 2021-04-14:
By the way, is there any good way to detect Vsync on Linux platforms? I couldn't find it.

jwinarske, 2021-04-14:
For Wayland refer to this:
https://lists.freedesktop.org/archives/wayland-devel/2016-March/027465.html

jwinarske, 2021-04-14:
For DRM this is a good reference:
https://github.com/dvdhrm/docs/blob/master/drm-howto/modeset-vsync.c

HidenoriMatsubayashi, 2021-04-16:
In terms of Wayland backend, I think `wp_presentation` is better than `wl_callback_listener`. We can get a refresh rate by using it. However, we don't call `wl_surface_commit` because of using `eglSwapBuffers` which is called from Flutter Engine. So, `wp_presentation` events doesn't happen continuously every frame.

HidenoriMatsubayashi, 2021-05-24:
I'm implementing to add vsync support for Wayland, but I separed this issue into x11, drm, and wayland respectively.

commit 09458fc2 (2021-05-26): Add vsync wayland support (#140)

Closed #68
closed 2021-05-26 by HidenoriMatsubayashi