sony/flutter-embedded-linux #79: DRM-GBM backend binaries load Wayland libraries?

HidenoriMatsubayashi, 2021-04-17:
The DRM-GBM binaries appear to contain the Wayland libraries. It works with RPI4, but it will get a segmentation fault with i.mx8m devices.

cc: @Mai-Matsuura 

## Related issues
- https://github.com/sony/flutter-embedded-linux/issues/4

HidenoriMatsubayashi, 2021-04-17:
I'm using the BSP built for Wayland, that may be the cause.

HidenoriMatsubayashi, 2021-04-17:
gbm.h needs to be included before context_egl.h is first included, but I feel like not following that. We also need to refactor code files.

LefixBlue, 2021-04-21:
@HidenoriMatsubayashi

Not the following statement from i.MX Yocto Project User's Guide regarding available distros:

> • fsl-imx-wayland - Wayland weston graphics.
• fsl-imx-xwayland - Wayland graphics and X11. X11 applications using EGL are not supported.
• fsl-imx-fb - Frame Buffer graphics - no X11 or Wayland. Frame Buffer is not supported on i.MX 8.

I would understand if then only fsl-imx-wayland supports EGL or maybe none of them?!

Maybe it helps.

HidenoriMatsubayashi, 2021-04-21:
Thank you for your comment. Yes, I knew that. I'm using fsl-imx-wayland now. However, I haven't known the cause of the following error yet :cry: `glmark2-es2-drm` works on i.MX8M mini.

```
Thread 1 "flutter-drm-gbm" received signal SIGSEGV, Segmentation fault.
0x0000fffff6fe3eb4 in __GI___pthread_mutex_lock (mutex=0xe9) at pthread_mutex_lock.c:67
67      pthread_mutex_lock.c: No such file or directory.
(gdb) backtrace
#0  0x0000fffff6fe3eb4 in __GI___pthread_mutex_lock (mutex=0xe9) at pthread_mutex_lock.c:67
#1  0x0000fffff6c6c340 in wl_proxy_create_wrapper () from /usr/lib/libwayland-client.so.0
#2  0x0000fffff7f448e0 in ?? () from /usr/lib/libEGL.so.1
#3  0x0000fffff7f3721c in eglInitialize () from /usr/lib/libEGL.so.1
```

LefixBlue, 2021-04-21:
Do you build the drm-gbm-backend on the target or cross via yocto?

I just wonder how you made it built through without errors ( note my comment in #4 )
Maybe it is still something wrong on my side.

HidenoriMatsubayashi, 2021-04-22:
In conclusion, I'm afraid we can't run Flutter with drm-gbm backend on i.mx8m platforms. We need to use Wayland backend on i.mx8m platforms. See: https://github.com/sony/flutter-embedded-linux/issues/4#issuecomment-824419854

closed 2021-04-22 by HidenoriMatsubayashi