sony/flutter-embedded-linux #87: Segmentation faut with flutter-client

topolittle, 2021-04-20:
I tried flutter-client (wayland) with no luck.

Environment:
Board: NXP imx8mm (lpddr4)
BSP/Kernel: 5.4.70-2.3.0
Distro: fsl-imx-wayland
Flutter-engine: 1.26-rc17 compiled to aarch64 from the official repository. Config switches:--unoptimized --embedder-for-target --enable-fontconfig --full-dart-sdk --enable-skshaper--runtime-mode debug

At first I got the following error:
```
root@imx8mmlpddr4evk:~# flutter-client ./bundle 
embedder.cc (1933): 'FlutterEnginePostDartObject' returned 'kInvalidArguments'. Engine not running.
embedder.cc (930): 'FlutterEngineInitialize' returned 'kInvalidArguments'. Not running in AOT mode but could not resolve the kernel binary.
[ERROR][flutter_linuxes_engine.cc(221)] Failed to start Flutter engine: error 2
Failed to create view controller.
```
After some debugging with strace, we can see that the executable tries to load the bundle from /usr/bin:
`newfstatat(AT_FDCWD, "/usr/bin/./bundle/data/flutter_assets/kernel_blob.bin", 0xffffc0d18640, 0) = -1 ENOENT (No such file or directory)`
```
So after copying the demo app bundel in /usr/bin, I got the following error:
root@imx8mmlpddr4evk:~# flutter-client ./bundle
embedder.cc (1933): 'FlutterEnginePostDartObject' returned 'kInvalidArguments'. Engine not running.
flutter: Observatory listening on http://127.0.0.1:33095/EMMK_h2PrrA=/
[ 1633.147478] audit: type=1701 audit(1633.140:3): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=457 comm="flutter-client" exe="/usr/bin/flutter-client" sig=11 res=1
Segmentation fault
```

I also tried the drm-egl version with no luck:
```
root@imx8mmlpddr4evk:~# export FLUTTER_DRM_DEVICE=/dev/dri/card1
root@imx8mmlpddr4evk:~# flutter-drm-eglstream-backend --help
[ERROR][environment_egl_drm_eglst[  141.702444] audit: type=1701 audit(387.168:3): auid=4294967295 uid=0 gid=0 ses=4294967295 pid=384 comm="flutter-drm-egl" exe="/usr/bin/flutter-drm-eglstream-backend" sig=11 res=1
ream.cc(15)] Failed to set extension function pointers
[ERROR][context_egl.cc(31)] Failed to choose EGL surface config: eglGetError: EGL_BAD_DISPLAY
[ERROR][context_egl_drm_eglstream.cc(20)] Failed to set extension function pointers
Segmentation fault
```
My primary target is Wayland.

HidenoriMatsubayashi, 2021-04-20:
> embedder.cc (930): 'FlutterEngineInitialize' returned 'kInvalidArguments'. Not running in AOT mode but could not resolve the kernel binary.

> embedder.cc (1933): 'FlutterEnginePostDartObject' returned 'kInvalidArguments'. Engine not running.

According to this log, I look you are using Flutter Engine (libflutter_ngine.so) with debug mode, but you don't use the bundle built with debug mode. Could you check your way to build a flutter app?

```
$ flutter build linux build --debug
```

topolittle, 2021-04-22:
The problem was caused by my version of flutter-engine.so that was not the same version on the target as the SDK version used to compile the app. Once I modified my flutter-engine recipe in Yocto to get the the same SHA version as the SDK used to build the Flutter app, it just worked!

Note: I still have to copy the bundle to /usr/bin to get it working.

closed 2021-05-13 by HidenoriMatsubayashi