sony/flutter-embedded-linux #105: Typo in trace output for wayland app

psstoyanov, 2021-05-07:
There is a small typo on `Wayland` on this trace output:
https://github.com/sony/flutter-embedded-linux/blob/9b2c62b1cd351cab8e198bbe3425a23d8a9e8f9b/src/flutter/shell/platform/linux_embedded/window/linuxes_window_wayland.cc#L658

closed 2021-05-08 by HidenoriMatsubayashi

HidenoriMatsubayashi, 2021-05-08:
@psstoyanov Thank you for your report! By the way, we are getting ready to accept PRs from external. Please wait..

psstoyanov, 2021-05-12:
I thought I was imagining it at first but taking another look does show another typo on the same line :smile:  :
https://github.com/sony/flutter-embedded-linux/commit/fa3f8ad8cc564f541b1d12e05b16240ee92eee4c#diff-ec3ba2cc23d0434fa75e6fa5021cc904963806c541d645f2f1ee2db4ee4566ceR658:
`Created the Waylnad surface:` 

PRs from external will be nice indeed :slightly_smiling_face:

HidenoriMatsubayashi, 2021-05-13:
Thank you again... I fixed https://github.com/sony/flutter-embedded-linux/pull/113

commit fa3f8ad8 (2021-05-08): Fix typo in log message for Wayland (#106)

Close #105