sony/flutter-embedded-linux #108: On-screen keyboard doesn't work

HidenoriMatsubayashi, 2021-05-09:
Currently, the virtual keyboard input is only enabled when running as weston-desktop-shell. Reported from: https://twitter.com/PakoSStoyanov/status/1391034963030597632

HidenoriMatsubayashi, 2021-05-10:
I fixed the issue. But the virtual keyboard for Wayland is default disabled in `examples/flutter-wayland-client`.

psstoyanov, 2021-05-10:
Amazing, @HidenoriMatsubayashi ! 

Tonight, I will try with a few desktop environments and see how it will go.

Right now, I don't see a change on Plasma Mobile with the virtual keyboard enabled in `examples/flutter-wayland-client/cmake/user_config.cmake` (confirmed that `text-input-unstable-v1` is indeed built with the Wayland client). I don't know if this is specific to Plasma Mobile, my particular setup or something else entirely.

HidenoriMatsubayashi, 2021-05-10:
> Right now, I don't see a change on Plasma Mobile with the virtual keyboard enabled in examples/flutter-wayland-client/cmake/user_config.cmake

Did you use the latest version?

psstoyanov, 2021-05-10:
Yes, I'm using the latest commit: `commit d3d78896167cf2a7ccf10cf8e95b07f6a7e1418a`. But I've only tried on Plasma Mobile for now - will gather more information with Phosh and Gnome later today at least.

HidenoriMatsubayashi, 2021-05-10:
Thank you so much!

HidenoriMatsubayashi, 2021-05-11:
I'm using weston 8.0.0 on Ubuntu 20.04. According to your comment, I think it's better to switch from v1 to v3.

psstoyanov, 2021-05-11:
Found this ticket for`wlroots` and there is some interesting information about v2 and v3:
https://github.com/swaywm/wlroots/issues/717 
But this is outside of the current goal. 

[Bhushan Shah](https://twitter.com/BhushanNShah) was kind enough to give me a [few explanations](https://matrix.to/#/!nitCweDuNxEMsoekiZ:matrix.org/$162071727827683QFXHj:kde.org?via=matrix.org&via=kde.org&via=tchncs.de) about Qt and `text-input-unstable` on Plasma Mobile's Matrix [channel](https://matrix.to/#/#plasmamobile:matrix.org):
```
text-input-v3 protocol is supported by kwin compositor
And also phosh
Now apps can choose to support it for instance in case of flutter apps
But qt apps will use v2
And that is also supported by plasma mobile
Basic TLDR is, if you want to use it from flutter v3 (and potentially v4 when it comes out *sigh*) is way to go.
You don't need to wait for qt to support it
It needs to be supported from compositor
Which Kwin and wlroots does
```

The TLDR line sums it up nicely:
- `Basic TLDR is, if you want to use it from flutter v3 (and potentially v4 when it comes out *sigh*) is way to go.`
With v3, the wayland client will have widest compatibility.

I don't know how this would affect Weston though.

HidenoriMatsubayashi, 2021-05-14:
@psstoyanov 

Thank you for your investigation.

> With v3, the wayland client will have widest compatibility.

Yes, I'll try to switch from v1 to v3 and test on Weston.

HidenoriMatsubayashi, 2021-05-15:
Unfortunately, it seems Weston doesn't still support v3.
https://gitlab.freedesktop.org/wayland/weston/-/merge_requests/150

So, in embedded use cases, we cannot delete v1. We need to support both v1 and v3.

psstoyanov, 2021-05-16:
That's unfortunate. It would've been great to avoid the additional complexity of implementing two different versions.

HidenoriMatsubayashi, 2021-05-17:
@psstoyanov 

I added text-input v3 support (https://github.com/sony/flutter-embedded-linux/pull/120). I've confirmed that the on-screen keyboard wroks on GNOME Wayland (Ubuntu 21). Could you please try again?

psstoyanov, 2021-05-17:
Initial results with Plasma Mobile are great! The keyboard is popping up as expected.

Will check with Phosh and Gnome 40 later today.

HidenoriMatsubayashi, 2021-05-17:
Great! Thanks.

psstoyanov, 2021-05-20:
I wasn't able to check with Gnome 40 as it relegates to the X11 session on Manjaro for some reason. PostmarketOS does default to Wayland but I can't use it for this purpose (it's using musl instead of glibc - the library can't be linked).

The keyboard didn't show up the OSK on Manjaro Lomiri and I'm not entirely sure why. The following matrix doesn't list any of the input versions for Mir so this could be the reason why: https://gitlab.freedesktop.org/wayland/wayland-protocols/-/issues/39#implementation-matrix
It is in Alpha state outside of UBports so I don't put much weight into it. I will try to gather more information nonetheless.

Phosh and Plasma Mobile open up the onscreen keyboard so that is great. 

Unlike on Android/ iOS, the TextInput was obscured by the keyboard. I do not know how this is supposed to function to avoid the input field from being obscured. [Bhushan](https://twitter.com/BhushanNShah) suggested `input_area` and `cursor_rectangle` as possible solutions as different compositors deal with this differently. I believe `cursor_rectangle` refers to [set_cursor_rectangle](https://pywayland.readthedocs.io/en/v0.4.1/module/protocol/text_input_unstable_v3.html?highlight=cursor_rectangle#pywayland.protocol.text_input_unstable_v3.ZwpTextInputV3.set_cursor_rectangle).
Probably it's better to leave this complication for another ticket?

HidenoriMatsubayashi, 2021-05-20:
Thank you.

> Bhushan suggested input_area and cursor_rectangle as possible solutions as different compositors deal with this differently. I believe cursor_rectangle refers to set_cursor_rectangle.

Probably, I think so, too.

> Probably it's better to leave this complication for another ticket?

Yes, please.

psstoyanov, 2021-05-20:
Some additional observations:
If the user switches from the Flutter Wayland application with the onscreen keyboard opened, the keyboard will behave as expected (close or remain opened). 
However when the user returns back to the Flutter app, the text field appears to have the focus (cursor is blinking) but the onscreen keyboard will be closed. 
To bring the onscreen keyboard again - lose focus from the input field and tap on it again or focus on a different input field.

HidenoriMatsubayashi, 2021-05-20:
> However when the user returns back to the Flutter app, the text field appears to have the focus (cursor is blinking) but the onscreen keyboard will be closed. To bring the onscreen keyboard again - lose focus from the input field and tap on it again or focus on a different input field.

Thank you! I'll check it out later.

psstoyanov, 2021-05-20:
Possibly a better example as of when this could happen within the context of embedded device would be when the user interacts with a system bar - adjusting the volume, opening the list of applications, notifications, etc

HidenoriMatsubayashi, 2021-05-21:
> However when the user returns back to the Flutter app, the text field appears to have the focus (cursor is blinking) but the onscreen keyboard will be closed. To bring the onscreen keyboard again - lose focus from the input field and tap on it again or focus on a different input field.

Fixed: https://github.com/sony/flutter-embedded-linux/pull/129

psstoyanov, 2021-05-21:
Just made a quick check and I don't see a difference in the behaviour with Wayfire. I will check on Plasma Mobile afterwards with a debug flutter engine. Hopefully it will produce some useful logs.

Is #129 dependent on #131 by any chance?

HidenoriMatsubayashi, 2021-05-21:
Thank you.

I've tested on Ubuntu 21.04 (GNOME wayland)...

> Is #129 dependent on #131 by any chance?

No, they are irrelevant.

psstoyanov, 2021-05-21:
Hm, this is odd. On Plasma Mobile I can observe similar behaviour.

If you can, could you check how this example  would behave with Gnome Wayland?
<details>
<summary> Example with 2 TextFields </summary>

```
import 'package:flutter/material.dart';

void main() {
  runApp(MyApp());
}

class MyApp extends StatelessWidget {
  @override
  Widget build(BuildContext context) {
    return MaterialApp(
      theme: ThemeData.dark(),
      debugShowCheckedModeBanner: false,
      home: Scaffold(
        body: MyWidget(),
      ),
    );
  }
}

class MyWidget extends StatelessWidget {
  @override
  Widget build(BuildContext context) {
    return Column(
      mainAxisAlignment: MainAxisAlignment.center,
      children: <Widget>[
        Expanded(
          child: Container(),
        ),
        Text('Example with TextField at the bottom'),
        TextField(
          decoration: InputDecoration(
            border: OutlineInputBorder(),
            hintText: "Enter text here",
          ),
        ),
        Expanded(
          child: Container(),
        ),
        TextField(
          decoration: InputDecoration(
            border: OutlineInputBorder(),
            hintText: "Enter text here",
          ),
        ),
      ],
    );
  }
}
```

</details>

HidenoriMatsubayashi, 2021-05-22:
It works like this.

https://user-images.githubusercontent.com/62131389/119209773-98de9a80-bae3-11eb-892b-1298141f76e4.mov

psstoyanov, 2021-05-25:
Sorry for taking awhile - currently I'm making sure I can use maliit with Plasma to open on my x64 desktop or the Pinebook Pro. Also getting OBS working on the Wayland system with Pipewire (the release candidates of 27.0 seem to work well for now).

This should allow me to make comparison screencaptures on the various environments.

HidenoriMatsubayashi, 2021-05-25:
> Sorry for taking awhile

Don't mind! Your help is always helpful for us.

liangqi, 2021-06-10:
> Of course. It works like this.
> 
>  sample.mov

Which Wayland compositor and on screen keyboard are you using? Thanks.

HidenoriMatsubayashi, 2021-06-11:
I often use GNOME on Ubuntu 21 desktop for tests.

liangqi, 2021-06-11:
I think I find it, Universal Access->Screen Keyboard in GNOME. But I haven't managed to get it work together with other IM engine(for example, ibus) yet.

HidenoriMatsubayashi, 2021-06-11:
> I think I find it, Universal Access->Screen Keyboard in GNOME.

Yes, that's right.

psstoyanov, 2022-01-11:
I have been away for some time. Did something happen with the text input?
It seems broken on Phosh, PlasmaMobile and Plasma Wayland atm. I ought to check if the input call is being executed :thinking:  ( `WAYLAND_DEBUG` + looking for focus/ enter call maybe?)

If I can, I will check on Ubuntu later this week (as soon as I find a spare drive for the purpose).

Preferred Weston version was 8, right?

HidenoriMatsubayashi, 2022-01-12:
No changes. 

> Preferred Weston version was 8, right?

Yes, version 8 or 9 is preferred.

psstoyanov, 2022-01-12:
Alright, so the change was on my side on the devices:
- Can't pass `-k` ([virtualkeyboard](https://github.com/sony/flutter-embedded-linux/wiki/How-to-run-Flutter-apps#command-options) option) when using [flutter-elinux](https://github.com/sony/flutter-elinux)
That appears to be outside of the requirements however and would be best suited for a ticket on that project (not sure how it ).

@liangqi , I'm not familiar with ibus - I recall reading that the setup for it wayland use. I guess your goal is to use it for complex language support. Possibly take a look at fcitx5.
Unfortunately, onscreen keyboards on Wayland is messy :expressionless: 

Plasma Mobile is throwing the following error and the app crashes however I haven't spent much time trying to figure why this happens:
```
xdg_surface@19: error -1: invalid window geometry size (360x-47)
```
Not observable on Plasma Wayland (desktop) or Phosh. 
Could be caused by something else entirely (sometimes the crash doesn't trigger - possibly because Plasma Mobile is forcing a resize?)

@HidenoriMatsubayashi , this ticket appears to have too wide of a scope. Would it be better to rename it as ` On-screen keyboard doesn't open: text_input_v1 / text_input_v3  ` and close it?

HidenoriMatsubayashi, 2022-01-14:
@psstoyanov 

> this ticket appears to have too wide of a scope. Would it be better to rename it as On-screen keyboard doesn't open: text_input_v1 / text_input_v3 and close it?

Thanks. Yes, I agree with you.

psstoyanov, 2022-01-21:
@HidenoriMatsubayashi , the ticket is created by you even if my tweet was the original cause of it :slightly_smiling_face: 
Can't really close it myself

HidenoriMatsubayashi, 2022-01-24:
@psstoyanov Thanks a lot.

closed 2022-01-24 by HidenoriMatsubayashi