sony/flutter-embedded-linux #127: Touch gesture doesn't work correctly for Wayland

HidenoriMatsubayashi, 2021-05-21:
Thank you for reporing a bug. @jcviau 
See: https://github.com/sony/flutter-embedded-linux/issues/34#issuecomment-845149819

For us, the **main problem** is related to the touch gesture. 
- Scroll/drag continue randomly once I remove my finger (lists and 2D transformation when moving the object)
- Multi-touch isnt working for rotation, pinch zoom/pan (tested in 2D transformation page). It looks like it open a console when we try those gesture.

Other issues we saw:
**Mouse-only issue**
- A few operation (x axis motion, 2d transformation) raised errors in the console: _Unsupported cursor: Forbidden_
- The snackbar control raise an exception when clicking the action text on the snackbar

**Touch & Mouse issue**
- Bottom App Bar (with Notch) is displayed correctly but show errors in the console: _some draw get error and skipped during validation_ 
- Segmented control is displayed correctly but show errors in the console: _some draw get error and skipped during validation_ 
- Bottom navigation with selected Label can raise an exception: _Unhandled Exception: Null check operator used on a null value_
- Any click in the Shrine page raise an exception.
In Debug: _Cannot hit test a render box that has never been laid out._
In Release: _The method 'createTicker' was called on null._

Note: some animation are sluggish on our board, per example the Context Menu.

HidenoriMatsubayashi, 2021-05-24:
@jcviau Could you let me make sure of a couple of things? We'll do debugging from now.

1. You are using i.MX 8M Mini with wayland, right?
2. Will the above problems occur when I run the gallery app?
3. Does a particular operation on the gallery app always cause a problem?

jcviau, 2021-05-24:
@HidenoriMatsubayashi 
1. Yes, using i.MX 8M mini with Wayland and Weston 9 from NXP.
2. Should occur, me and a colleague had the same behavior on our setup. We started the application with --fullscreen argument.
3. All the problem listed occur everytime except for the Bottom navigation with selected Label where we had to change tab a few times before it occurs. 

Lots of those issues seems to happens on the bottom of the application. Also, I saw that the bottom seems cropped a little bit so maybe it's related.

Our setup:
- Screen: [11.6 Inch ](https://neewer.com/products/video-camera-supports-10095891)
- Board [i.MX 8M Mini LPD4 EVK ](https://www.nxp.com/design/development-boards/i-mx-evaluation-and-development-boards/evaluation-kit-for-the-i-mx-8m-mini-applications-processor:8MMINILPD4-EVK)(https://www.nxp.com/design/development-boards/i-mx-evaluation-and-development-boards/evaluation-kit-for-the-i-mx-8m-mini-applications-processor:8MMINILPD4-EVK)

HidenoriMatsubayashi, 2021-06-09:
@jcviau 

I report a progress.

> Scroll/drag continue randomly once I remove my finger (lists and 2D transformation when moving the object)

Unfortunately, we haven't reproduced it yet.

> Multi-touch isnt working for rotation, pinch zoom/pan (tested in 2D transformation page). It looks like it open a console when we try those gesture.

> A few operation (x axis motion, 2d transformation) raised errors in the console: Unsupported cursor: Forbidden

Unfortunately, we haven't reproduced it yet.

> The snackbar control raise an exception when clicking the action text on the snackbar

We could reproduce it, and it seems to be the issues of the Flutter framework or Flutter engine. I reported the issues. See: https://github.com/flutter/flutter/issues/84241

> Bottom App Bar (with Notch) is displayed correctly but show errors in the console: some draw get error and skipped during validation

Unfortunately, we haven't reproduced it yet.

> Segmented control is displayed correctly but show errors in the console: some draw get error and skipped during validation

Unfortunately, we haven't reproduced it yet.

> Bottom navigation with selected Label can raise an exception: Unhandled Exception: Null check operator used on a null value

Unfortunately, we haven't reproduced it yet.

> Any click in the Shrine page raise an exception.

Unfortunately, we haven't reproduced it yet.

anyuliu, 2021-09-07:
Hi ,
I also encountered the following problem

Scroll/drag continue randomly once I remove my finger (lists and 2D transformation when moving the object)

After debugging, it is found that the touch event timestamp may be abnormal

The problem disappeared when I updated the timestamp

```
diff --git a/src/flutter/shell/platform/linux_embedded/flutter_elinux_view.cc b/src/flutter/shell/platform/linux_embedded/flutter_elinux_view.cc
index 869439a..112c11c 100644
--- a/src/flutter/shell/platform/linux_embedded/flutter_elinux_view.cc
+++ b/src/flutter/shell/platform/linux_embedded/flutter_elinux_view.cc
@@ -103,6 +103,9 @@ void FlutterELinuxView::OnTouchDown(uint32_t time, int32_t id, double x,
   if (!point) {
     return;
   }
+  time = std::chrono::duration_cast < std::chrono::microseconds
+      > (std::chrono::high_resolution_clock::now().time_since_epoch()).count();
+
   point->event_mask = TouchEvent::kDown;
   point->x = x;
   point->y = y;
@@ -129,6 +132,9 @@ void FlutterELinuxView::OnTouchUp(uint32_t time, int32_t id) {
   if (!point) {
     return;
   }
+  time = std::chrono::duration_cast < std::chrono::microseconds
+      > (std::chrono::high_resolution_clock::now().time_since_epoch()).count();
+
   point->event_mask = TouchEvent::kUp;

   FlutterPointerEvent event = {
@@ -153,6 +159,9 @@ void FlutterELinuxView::OnTouchMotion(uint32_t time, int32_t id, double x,
   if (!point) {
     return;
   }
+  time = std::chrono::duration_cast < std::chrono::microseconds
+      > (std::chrono::high_resolution_clock::now().time_since_epoch()).count();
+
   point->event_mask = TouchEvent::kMotion;
   point->x = x;
   point->y = y;
```

HidenoriMatsubayashi, 2021-09-07:
@anyuliu 

Thank you for your report! The time-stamp for touch inputs uses the value that comes from a Wayland compositor as it is. Could you inform me of what value is sending in your environment?

anyuliu, 2021-09-07:
@HidenoriMatsubayashi 

The printed value is like this, it looks strange

```
[ERROR][OnTouchDown(106)] wayland time=84564
[ERROR][OnTouchDown(111)] current time=84574912
[ERROR][OnTouchDown(113)] FlutterEngineGetCurrentTime=84575006580
[ERROR][OnTouchMotion(173)] wayland time=84605
[ERROR][OnTouchMotion(176)] current time=84620796
[ERROR][OnTouchMotion(177)] FlutterEngineGetCurrentTime=84620850330
[ERROR][OnTouchMotion(173)] wayland time=84626
[ERROR][OnTouchMotion(176)] current time=84636368
[ERROR][OnTouchMotion(177)] FlutterEngineGetCurrentTime=84636423039
[ERROR][OnTouchMotion(173)] wayland time=84647
[ERROR][OnTouchMotion(176)] current time=84651990
[ERROR][OnTouchMotion(177)] FlutterEngineGetCurrentTime=84652046789
[ERROR][OnTouchMotion(173)] wayland time=84668
[ERROR][OnTouchMotion(176)] current time=84682568
[ERROR][OnTouchMotion(177)] FlutterEngineGetCurrentTime=84682623205
[ERROR][OnTouchMotion(173)] wayland time=84689
[ERROR][OnTouchMotion(176)] current time=84698536
[ERROR][OnTouchMotion(177)] FlutterEngineGetCurrentTime=84698597664
[ERROR][OnTouchMotion(173)] wayland time=84710
[ERROR][OnTouchMotion(176)] current time=84714030
[ERROR][OnTouchMotion(177)] FlutterEngineGetCurrentTime=84714086455
[ERROR][OnTouchUp(141)] wayland time=84731
[ERROR][OnTouchUp(145)] current time=84744752
[ERROR][OnTouchUp(147)] FlutterEngineGetCurrentTime=84744806622
```

HidenoriMatsubayashi, 2021-09-07:
@anyuliu 

Thanks a lot. The cause was the wrong unit of time. So, we need to multiply the value from wayland by 1000. I'll fix the bug tomorrow.

Wayland: milliseconds (https://wayland-book.com/seat/touch.html#touch-and-release)
Flutter Engine: microseconds (https://github.com/flutter/engine/blob/master/shell/platform/embedder/embedder.h#L651)

Thank you so much!

jcviau, 2022-01-07:
We did a quick validation and all the issues listed here seems to be fixed! Should close this issue. 

Thanks @anyuliu @HidenoriMatsubayashi

HidenoriMatsubayashi, 2022-01-10:
@jcviau Thank you so much.

closed 2022-01-10 by HidenoriMatsubayashi