sony/flutter-embedded-linux #138: No rule to make target 'libflutter_engine.so', needed by 'flutter-client'

alfiedouglas-tewke, 2021-05-25:
I have unable to build the examples due to the following error:

```
$ cmake --build .
Scanning dependencies of target flutter-client
[  2%] Building CXX object CMakeFiles/flutter-client.dir/examples/flutter-wayland-client/flutter_window.cc.o
...
[ 81%] Building CXX object CMakeFiles/flutter-client.dir/src/flutter/shell/platform/common/engine_switches.cc.o
[ 84%] Building CXX object CMakeFiles/flutter-client.dir/src/flutter/shell/platform/common/incoming_message_dispatcher.cc.o
make[2]: *** No rule to make target 'libflutter_engine.so', needed by 'flutter-client'. Stop.
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/flutter-client.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

I have downloaded `libflutter_engine.so` and copied it into `/usr/lib`

alfiedouglas-tewke, 2021-05-25:
I was unaware I needed to copy `libflutter_engine.so` into the build directory.

closed 2021-05-25 by alfiedouglas-tewke

HidenoriMatsubayashi, 2021-05-25:
Don't mind! And thank you for trying to use this software.