sony/flutter-embedded-linux #139: Run with Wayland backend

alfiedouglas-tewke, 2021-05-25:
After following the [docs](https://github.com/sony/flutter-embedded-linux/blob/master/doc/README.md), I have been trying to run with Wayland backend, however I get the error which I have been unable to resolve:

```
./flutter-client --bundle=./sample/build/linux/x64/release/bundle/
embedder.cc (939): 'FlutterEngineInitialize' returned 'kInvalidArguments'. Not running in AOT mode but could not resolve the kernel binary.
[ERROR][flutter_linuxes_engine.cc(221)] Failed to start Flutter engine: error 2
Failed to create view controller.
Failed to create a Flutter window.
```

Any help would be greatly appreciated! Thanks

jcviau, 2021-05-25:
@alfiedouglas-tewke looks like the issue mentionned here : https://github.com/sony/flutter-embedded-linux/issues/4#issuecomment-834348065

You could try to copy the bundle folder to /usr/bin/

jcviau, 2021-05-25:
@alfiedouglas-tewke maybe related to  https://github.com/sony/flutter-embedded-linux/issues/4#issuecomment-834348065. Similar issue (kInvalidArguments).

alfiedouglas-tewke, 2021-05-25:
@jcviau moving the bundle in to `/usr/bin` didn't fix it unfortunately, I still get the same error

HidenoriMatsubayashi, 2021-05-25:
@alfiedouglas-tewke 

> embedder.cc (939): 'FlutterEngineInitialize' returned 'kInvalidArguments'. Not running in AOT mode but could not resolve the kernel binary.

I guess you built `libflutter_engine.so` with **debug** mode. If so, you need a bundle that is built with debug mode.

How to build with debug mode:
```
$ flutter build linux --debug
```

How to run
```
./flutter-client --bundle=./sample/build/linux/x64/debug/bundle/
```

alfiedouglas-tewke, 2021-05-25:
@HidenoriMatsubayashi 

> I guess you built `libflutter_engine.so` with debug mode. If so, you need a bundle that is built with debug mode.

I downloaded `libflutter_engine.so` (with the master hash) from https://storage.googleapis.com/flutter_infra/flutter/FLUTTER_ENGINE/linux-x64/linux-x64-embedder.

I have just tried building the flutter app with `--debug`, and I now get this error when running:

```
$ ./flutter-client --bundle=./sample/build/linux/x64/debug/bundle
flutter: Observatory listening on http://127.0.0.1:38589/a-ZaU5K8tTc=/
[ERROR:flutter/shell/common/shell.cc(103)] Dart Error: Can't load Kernel binary: Invalid kernel binary format version.
[ERROR:flutter/runtime/dart_isolate.cc(171)] Could not prepare isolate.
[ERROR:flutter/runtime/runtime_controller.cc(415)] Could not create root isolate.
[ERROR:flutter/shell/common/shell.cc(588)] Could not launch engine with configuration.
```

Seems like it's related to https://github.com/sony/flutter-embedded-linux/issues/4#issuecomment-797028160

HidenoriMatsubayashi, 2021-05-25:
Could you make sure that the downloaded `libflutter_engine.so` version matches the Flutter SDK version?

Flutter SDK: ${path_to_flutter_sdk_install}/flutter/bin/internal/engine.version

Downloaded Engine version: check `FLUTTER_ENGINE`
 https://storage.googleapis.com/flutter_infra/flutter/FLUTTER_ENGINE/linux-x64/linux-x64-embedder

HidenoriMatsubayashi, 2021-05-25:
The current procedure is not easy for beginners, so we are currently considering making a tool.
See: https://github.com/sony/flutter-embedded-linux/issues/59

charafau, 2021-05-26:
@HidenoriMatsubayashi  unfortunetly I have the same problem, I replaced hash in my flutter installation:

```
➜  internal git:(dev) pwd 
/home/charafau/Utils/flutter/bin/internal
➜  internal git:(dev) cat engine.version 
26e217e6c3d487bb3c0cbf8a7edb378952a51d33
```
then downloaded engine with wget:

```
➜  flutter-embedded-linux git:(master) ✗ wget https://storage.googleapis.com/flutter_infra/flutter/26e217e6c3d487bb3c0cbf8a7edb378952a51d33/linux-x64/linux-x64-embedder
--2021-05-26 03:25:46--  https://storage.googleapis.com/flutter_infra/flutter/26e217e6c3d487bb3c0cbf8a7edb378952a51d33/linux-x64/linux-x64-embedder

```
my build folder:

```
➜  build git:(master) ✗ ls
CMakeCache.txt  CMakeFiles  cmake_install.cmake  libflutter_engine.so  Makefile

```
I build sample flutter app like so 
```
➜  sample git:(master) ✗ flutter build linux 
```

and run it:

```
 ➜  build git:(master) ✗ ./flutter-client --bundle=./sample/build/linux/x64/release/bundle
embedder.cc (982): 'FlutterEngineInitialize' returned 'kInvalidArguments'. Not running in AOT mode but could not resolve the kernel binary.
[ERROR][flutter_linuxes_engine.cc(221)] Failed to start Flutter engine: error 2
Failed to create view controller.
Failed to create a Flutter window.
```

maybe something on server has changed? when I caculate that zip file instead of my sum ( 26e217e6c3d487bb3c0cbf8a7edb378952a51d33 )

I get:

```
➜  flutter-embedded-linux git:(master) ✗ sha1sum linux-x64-embedder 
02de140e89b21eb39ad309d0d85eaaef13e76d12  linux-x64-embedder
➜  flutter-embedded-linux git:(master) ✗ sha1sum libflutter_engine.so 
44b401b44f34b25a5a13cdeef2f3e36be24a1aaa  libflutter_engine.so
```

unless it's not sha1sum but that's my only idea

I run ubuntu 21.04 and try to build standalone wayland application

HidenoriMatsubayashi, 2021-05-26:
@charafau 

Thank you for your comment.

> https://storage.googleapis.com/flutter_infra/flutter/26e217e6c3d487bb3c0cbf8a7edb378952a51d33/linux-x64/linux-x64-embedder

This library (`libflutter_engine.so`) is for debug mode.

> ➜  sample git:(master) ✗ flutter build linux 

This command is for release build. So, you need to use the following command.

```
$ flutter build linux --debug
$ ./flutter-client --bundle=./sample/build/linux/x64/debug/bundle
```

HidenoriMatsubayashi, 2021-05-26:
> tried again today and it works

Good, good! You're welcom.

alfiedouglas-tewke, 2021-05-26:
Hi @HidenoriMatsubayashi,

> Could you make sure that the downloaded libflutter_engine.so version matches the Flutter SDK version?

Yeah I have just retried ensuring the same version is used. Here is the process and error:

```
$ mkdir build
$ cd build
build$ which flutter
/opt/flutter/bin/flutter
build$ cat /opt/flutter/bin/internal/engine.version 
b1385c0df12549b9dce7ce3fee361bd4898f49ae
build$ curl -O https://storage.googleapis.com/flutter_infra/flutter/b1385c0df12549b9dce7ce3fee361bd4898f49ae/linux-x64/linux-x64-embedder
build$ unzip linux-x64-embedder 
Archive:  linux-x64-embedder
  inflating: flutter_embedder.h      
  inflating: libflutter_engine.so 
build$
build$ cmake -DUSER_PROJECT_PATH=examples/flutter-wayland-client ..
[...]
build$ make
[...]
build$
build$ flutter create sample
[...]
build$ cd sample
build/sample$ flutter build linux --debug
build/sample$ cd ..
build$
build$ ./flutter-client --bundle=./sample/build/linux/x64/debug/bundle/
flutter: Observatory listening on http://127.0.0.1:41017/uNq9rCHGCQQ=/
[ERROR:flutter/shell/common/shell.cc(103)] Dart Error: Can't load Kernel binary: Invalid kernel binary format version.
[ERROR:flutter/runtime/dart_isolate.cc(171)] Could not prepare isolate.
[ERROR:flutter/runtime/runtime_controller.cc(415)] Could not create root isolate.
[ERROR:flutter/shell/common/shell.cc(588)] Could not launch engine with configuration.
```

I have also tried with building the flutter app with `--release` and the error I get is the original one:
```
build$ ./flutter-client --bundle=./sample/build/linux/x64/release/bundle/
embedder.cc (939): 'FlutterEngineInitialize' returned 'kInvalidArguments'. Not running in AOT mode but could not resolve the kernel binary.
[ERROR][flutter_linuxes_engine.cc(221)] Failed to start Flutter engine: error 2
Failed to create view controller.
Failed to create a Flutter window.
```

charafau, 2021-05-26:
@alfiedouglas-tewke  for your debug build, do yo have libflutter_engine.so in you LD_LIBRARY_PATH ? 

try to run this inside your build directory

`export LD_LIBRARY_PATH=.`

alfiedouglas-tewke, 2021-05-26:
@charafau that was it! Thanks a lot!

closed 2021-05-26 by alfiedouglas-tewke