sony/flutter-embedded-linux #157: Add external texture plugin for GPU texture

HidenoriMatsubayashi, 2021-06-02:
The current external texture plugin which is the same with Flutter desktop for Windows supports pixel buffer only. However, its performance is not good. So we need to add a texture plugin with a GPU texture buffer.

charafau, 2021-06-14:
related issue and PR (for :
https://github.com/flutter/flutter/issues/83693
https://github.com/flutter/engine/pull/26562

implementation in tizen engine:

https://github.com/flutter-tizen/engine/pull/86

HidenoriMatsubayashi, 2021-06-14:
Thank you for sharing. I guess flutter/engine#26562 is the same implementation with flutter-tizen. I'm looking forward to it will be merged. After that, I'll copy and merge it to our repo.

charafau, 2021-06-14:
yes, I think you're right. the reason why I linked tizen one is because in discussion, dev actually shows how to use this code :)

swift-kim, 2022-05-12:
Update: https://github.com/flutter/engine/pull/26562#issuecomment-1114156009

talksik, 2023-10-08:
Hey @HidenoriMatsubayashi, isn't this thread solved by your work [here](https://github.com/sony/flutter-embedded-linux/issues/247) with DMA buffers? Or is that different than external texture plugin.

Is using the DMA buffer usage that is current implemented in `video_player` ensure hardware accelerated rendering? Or is it just improving the memory inefficiency problem compared to PixelBuffer?

Thank you! Been trying to figure out how to improve performance here further. I have gstreamer elements with hardware accelerated decoding, but now need to ensure that rendering is done with hardare acceleration.

HidenoriMatsubayashi, 2023-10-08:
> isn't this thread solved by DMA buffers? Or is that different than external texture plugin.

No, it's different from dmabuf in video_player. The task in this issue is that porting https://github.com/flutter/engine/pull/26840. Currently, we haven't implemented `kFlutterDesktopGpuSurfaceTexture` yet.
See https://github.com/sony/flutter-embedded-linux/blob/master/src/flutter/shell/platform/linux_embedded/flutter_elinux_texture_registrar.cc#L51C45-L51C45

talksik, 2023-10-09:
Thanks @HidenoriMatsubayashi for the detailed explanation!

I will research this further to see if I can make a dent. 👍🏾

barribarrier, 2024-09-21:
Hi @HidenoriMatsubayashi 

Has there been any progress on this? I wanted to do harware accelerated video rendering using `video_player` too.

HidenoriMatsubayashi, 2024-09-21:
unfortunately, there is no progress. So it would be great if you could create a patch for this :)

barribarrier, 2024-09-21:
@HidenoriMatsubayashi Got it.
Was there anything wrong with this or is it just left unfinished? https://github.com/HidenoriMatsubayashi/flutter-embedded-linux/commit/d7750d3bfe81c02a860219e676c54675ab037923

HidenoriMatsubayashi, 2024-09-21:
I don't remember, but it’s unfinished and old.