sony/flutter-embedded-linux #166: Flutter engine support for 32 bit ARM

kushflutter, 2021-06-11:
@HidenoriMatsubayashi Do we have support available for ARM 32 bit?
Do we have plans to support it?
If i need to do by myself, any leads that can help?

HidenoriMatsubayashi, 2021-06-11:
Technically, yes. This software (flutter-embedded-linux) doesn't have any CPU architecture-dependent implementations. However, there are some issues to ARM 32bit (armhf, ARMv7) support of the Flutter Engine.

- https://github.com/flutter/flutter/issues/83765
- https://github.com/flutter/flutter/issues/84133
- https://github.com/flutter/flutter/issues/74322
- https://github.com/flutter/flutter/issues/75823

chinmaygarde, 2021-06-14:
Is this armv7s or armv6?

kushflutter, 2021-06-14:
> Is this armv7s or armv6?

@chinmaygarde - This is for armv7

HidenoriMatsubayashi, 2022-08-07:
Closing as known issues have been fixed.

closed 2022-08-07 by HidenoriMatsubayashi