sony/flutter-embedded-linux #179: Can't load Kernel binary: Invalid kernel binary format version.

dongzhangsun, 2021-06-21:
hi,
This error occurs when I cross compile the flitter app（host:linux x64, target:linux arm64）。

${FLUTTER_SDK}/bin/cache/artifacts/engine/linux-arm64-release/clang_x64/gen_snapshot \
  --deterministic \
  --snapshot_kind=app-aot-elf \
  --elf=.dart_tool/flutter_build/flutter-embedded-linux/libapp.so \
  --strip \
  .dart_tool/flutter_build/flutter-embedded-linux/app.dill

Can't load Kernel binary: Invalid kernel binary format version.

HidenoriMatsubayashi, 2021-06-21:
Could you confirm that the version between the Flutter SDK version and Flutter engine which you built are matched?
 See also: https://github.com/sony/flutter-embedded-linux/wiki/Building-Flutter-apps#1-installing-flutter-sdk

dongzhangsun, 2021-06-22:
> Could you confirm that the version between the Flutter SDK version and Flutter engine which you built are matched?
> See also: https://github.com/sony/flutter-embedded-linux/wiki/Building-Flutter-apps#1-installing-flutter-sdk

Thank you. Everything is working right now！

HidenoriMatsubayashi, 2021-06-22:
Great. I'll close this issues.

closed 2021-06-22 by HidenoriMatsubayashi