sony/flutter-embedded-linux #181: flutter display is very blurry.

dongzhangsun, 2021-06-22:
My monitor is HDMI interface, resolution is 1920 × 1080, Now the display looks blurry（linux arm64）, but it is clear on my working computer(linux x64).

dongzhangsun, 2021-06-22:
[image]

HidenoriMatsubayashi, 2021-06-23:
Actually, this has been on my mind for some time. The same symptomatism seems to occur with flutter desktop for linux. However, the texts are clear with flutter desktop for mac.

HidenoriMatsubayashi, 2023-08-08:
I'll close this issue.

The flutter text’s default font size is 14 and I think it's small.
See https://madewithflutter.net/what-is-flutter-texts-default-font-size/

After changing the font size, it the text will be clear.

closed 2023-08-08 by HidenoriMatsubayashi