sony/flutter-embedded-linux #186: Failed to make the EGL context current: eglGetError: EGL_BAD_NATIVE_WINDOW

dongzhangsun, 2021-06-25:
This error occurs after the app has been running for some time.

[ERROR][elinux_egl_surface.cc(29)] Failed to make the EGL context current: eglGetError: EGL_BAD_NATIVE_WINDOW
[ERROR:flutter/shell/gpu/gpu_surface_gl.cc(110)] Could not make the context current to destroy the GrDirectContext resources.

dongzhangsun, 2021-06-25:
[ERROR][elinux_egl_surface.cc(38)] Failed to swap the EGL buffer: eglGetError: EGL_BAD_NATIVE_WINDOW
[ERROR][elinux_egl_surface.cc(29)] Failed to make the EGL context current: eglGetError: EGL_BAD_NATIVE_WINDOW
[ERROR:flutter/shell/gpu/gpu_surface_gl.cc(228)] Could not make the context current to acquire the frame.
[ERROR][elinux_egl_surface.cc(29)] Failed to make the EGL context current: eglGetError: EGL_BAD_NATIVE_WINDOW
[ERROR:flutter/shell/gpu/gpu_surface_gl.cc(110)] Could not make the context current to destroy the GrDirectContext resources.

HidenoriMatsubayashi, 2021-06-25:
Could you inform me of the version and the backend you are using? Wayland?

dongzhangsun, 2021-06-25:
> Could you inform me of the version, the backend, and the options you are using? Wayland?

flutter-embedded-linux  version is latest.
system: linux builroot (linux kernel 4.9)
backend:wayland
cmd:flutter-client  --bundle=/linux-embedded-arm64/release/bundle --width=1920 --height=1080

HidenoriMatsubayashi, 2021-06-25:
Thanks a lot. I'll test it on some arm64 devices next week.

dongzhangsun, 2021-06-25:
> Thanks a lot. I'll test it on some arm64 devices next week.

Now I found a rule. When I started the app, my mouse was not plugged into the board. After a while, I plugged in the usb mouse. So far it is normal, but when I start to move the mouse,The above error will occur.

HidenoriMatsubayashi, 2021-06-25:
Thanks!

Does this issue happen on your desktop (x64) Linux host? These days, I didn't test it on arm64 devices, but it works fine on Linux desktop (Ubuntu 20.04, x64).

dongzhangsun, 2021-06-25:
It works normally on my desktop computer(Manjaro linux x64).

dongzhangsun, 2021-06-25:
> Thanks!
> 
> Does this issue happen on your desktop (x64) Linux host? These days, I didn't test it on arm64 devices, but it works fine on Linux desktop (Ubuntu 20.04, x64).

It works normally on my desktop computer(Manjaro linux x64).

HidenoriMatsubayashi, 2021-06-25:
Thanks. I'll test it next week and share the results.

HidenoriMatsubayashi, 2021-07-01:
@dongzhangsun 
Unfortunately, I've tried to reproduce this issue, but I haven't reproduced it yet.

HidenoriMatsubayashi, 2023-08-11:
I'm not sure what devices you used, but I'll close this issue once since I haven't seen this issue on my arm64 devices.

closed 2023-08-11 by HidenoriMatsubayashi