sony/flutter-embedded-linux #211: Where is the file flutter-client?

M201370367, 2021-08-17:
For now, I can build flutter-linux-arm64 app sucessfully. But I  can not run it, because i can not find flutter-client command. 
Who can help answer the question?
Thanks a lot.

HidenoriMatsubayashi, 2021-08-17:
[Build for Wayland backend (Stand-alone Wayland app)](https://github.com/sony/flutter-embedded-linux/wiki/Building-Embedded-Linux-embedding-for-Flutter#build-for-wayland-backend-stand-alone-wayland-app) will help you. Also, it is more easy to use our [flutter-elinux](https://github.com/sony/flutter-elinux) that a non-official extension to the Flutter SDK to build and debug Flutter apps for Embedded Linux devices.

M201370367, 2021-08-18:
Thanks for your answer.

closed 2021-08-18 by M201370367