sony/flutter-embedded-linux #218: Simplify Installation of Flutter Engine Library & Cross Compilation

Taha-Firoz, 2021-09-06:
I've already used [flutter pi](https://github.com/ardera/flutter-pi) cross compilation was very straight forward and simple while also providing steps for both Windows and Linux. 

Going through the documentation for this project the installation of the flutter engine library is a lot more complex since it requires building it yourself. Compared to how the flutter pi dev has provided [pre compiled libraries for arm and arm64 platforms](https://github.com/ardera/flutter-engine-binaries-for-arm) while providing reasonable explanations for each step. 

Is there someway we can streamline the cross compilation process for Windows and Linux since that's going to be the case for a majority of embedded systems development use cases.

HidenoriMatsubayashi, 2021-09-06:
If you are OK to use a pre-built image, use https://github.com/sony/flutter-elinux. However, if you develop your real product, it would be better to build Flutter Engine and this embedder yourself using your target toolchain and use it. Does it make sense?

Taha-Firoz, 2021-09-10:
Yes it does make sense, I was going through setting up a Yocto build for the first time but since I'm very new to all of this and it seems very very complicated. I would like to help out by making a PR to maybe improve the documentation to make it more approachable for beginners, once I figure out how to make  a build myself.

In a very basic way the flutter-pi repository breaks down the entire process in a very simple and approachable way and I hope maybe we can break down the step for this project. I think focusing more on cross compilation is important since that's what any developer will be focusing on and keeping the Raspberry Pi 4 as a reference board is important since it runs ARM 64 bit and is the most accessible to beginners.
 
### Here's a reference to the flutter-pi build + deploy process documentation.

1. Find out the path to your flutter SDK. For me it's `C:\flutter`. (I'm on Windows)
2. Open terminal or commandline and `cd` into your app directory.
<details>

3. Build the asset bundle.
   ```
   flutter build bundle
   ```
4. Build the kernel snapshot. (Replace `my_app_name` with the name of your app)
    ```cmd
    C:\flutter\bin\cache\dart-sdk\bin\dart.exe ^
      C:\flutter\bin\cache\dart-sdk\bin\snapshots\frontend_server.dart.snapshot ^
      --sdk-root C:\flutter\bin\cache\artifacts\engine\common\flutter_patched_sdk_product ^
      --target=flutter ^
      --aot ^
      --tfa ^
      -Ddart.vm.product=true ^
      --packages .packages ^
      --output-dill build\kernel_snapshot.dill ^
      --verbose ^
      --depfile build\kernel_snapshot.d ^
      package:my_app_name/main.dart
    ```
5. Fetch the latest `gen_snapshot_linux_x64_release` I provide in the [engine binaries repo](https://github.com/ardera/flutter-engine-binaries-for-arm).
6. The following steps must be executed on a linux x64 machine. If you're on windows, you can use [WSL](https://docs.microsoft.com/de-de/windows/wsl/install-win10). If you're on macOS, you can use a linux VM.
7. Build the `app.so`. If you're building for _arm64_, you need to omit the `--sim-use-hardfp` flag.
    ```bash
    gen_snapshot_linux_x64_release \
      --deterministic \
      --snapshot_kind=app-aot-elf \
      --elf=build/flutter_assets/app.so \
      --strip \
      --sim-use-hardfp \
      build/kernel_snapshot.dill
    ```
8. Now you can switch to your normal OS again.
9. Upload the asset bundle and the `app.so` to your Raspberry Pi.
    ```bash
    rsync -a --info=progress2 ./build/flutter_assets/ pi@raspberrypi:/home/pi/my_app
    ```
    or
    ```
    scp -r ./build/flutter_assets/ pi@raspberrypi:/home/pi/my_app
    ```
10. You can now launch the app in release mode using `flutter-pi --release /home/pi/my_app`  

</details>

Your project goes into great depth of the entire build process, what components are needed, how those components are built etc. But due to the cross compilation process being much more involved and complicated it can make the entire process unapproachable. Even with the [flutter-elinux](https://github.com/sony/flutter-elinux) the quick start or documentation makes cross build seem like an unreasonable decision however it's one of the most important steps for anyone trying to get a build running on any ARM board.

HidenoriMatsubayashi, 2023-08-06:
I'll close this issue as there has been no updates for a long time. Thanks.

closed 2023-08-06 by HidenoriMatsubayashi

Krakert, 2024-09-16:
Im trying to get this running for the Raspberry Pi 4, but Im unable to get it to run. 

I find it hard to understand the different repos and what i need to do on which target. Is there a more simplified guide I could follow?

Im trying to use a prebuild embedder instead of builing one from scratch. But I cant get the `./flutter-client` tools to work.

psstoyanov, 2024-09-16:
There is no easy way to cross-compile. They all require some administrative knowledge for the platforms (both the one you are using and on your RPi target).

You can check the guides here:
- https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#cross-building-from-x64-to-arm64

Choose one and get familiar with it.

- QEMU + Docker is slow to run but easy to understand.
  - Isolated environment, QEMU is slow - might as well just flutter-elinux directly on the RPi with comparative compile speed.
- I can't speak about Yocto - haven't used it.
- Buildroot Toolchain is the slimmest however you **do** need to understand how the system paths work on your system.
  - Can mess your environment if you aren't careful

Easiest way to start:

 - Follow the [Quick Start](https://github.com/sony/flutter-elinux?tab=readme-ov-file#quick-start) guide directly on your Raspberry Pi 4