sony/flutter-embedded-linux #220: [Wayland] Slow frames

HidenoriMatsubayashi, 2021-09-09:
Obviously slow frames are more easily occurring with Wayland backend compared to X11 backend.

### Wayland
[image]

### X11
[image]

HidenoriMatsubayashi, 2021-11-09:
Reopened because of https://github.com/sony/flutter-embedded-linux/pull/235

Nagendra-Bankupalli, 2021-12-09:
@HidenoriMatsubayashi , I am testing flutter-elinux on ARM64 (imx8) platform, on wayland (drm-backend) , found the gpu and cpu utilisations are high for flutter-applications, would like to know  how to do better gpu profiling for flutter-elinux ? if this thread is not for gpu, please suggest me the right link,

HidenoriMatsubayashi, 2021-12-10:
> would like to know how to do better gpu profiling for flutter-elinux ?

I think BSP vendors usually provide the GPU profiling tools. I recommend you use it to analyze GPU performance. Whereas, if you want to analyze flutter apps performance or this embedder, you can use [DevTools](https://docs.flutter.dev/testing/debugging).

Nagendra-Bankupalli, 2021-12-10:
@HidenoriMatsubayashi , thank you for the inputs, sure will try devTools as well, but do we need to take any special care to enable GPU support while building flutter ?

HidenoriMatsubayashi, 2021-12-10:
Yes, but can you check whether GPU is enabled in your target platform using some tools like glmark2?

Nagendra-Bankupalli, 2021-12-10:
glmark2-es-wayland works fine on my HW!

B-Nagendra, 2022-10-12:
Hi on my platform glmark2-es-wayland works well, but flutter gallery app is slow, what could be the issue?

HidenoriMatsubayashi, 2022-10-12:
What device or board are you using?

B-Nagendra, 2022-10-12:
its ARM64 based system

HidenoriMatsubayashi, 2022-10-12:
Specifically?

B-Nagendra, 2022-10-12:
its custom hw, its TI based  (PowerVR + openGLES 3.1), powrVR supprted in flutter 2.5.0 onwards , (not sure about flutter-elinux)
flutter-elinux version is 
Flutter 3.3.3 • channel unknown • unknown source
Framework • revision 18a827f393 (2 weeks ago) • 2022-09-28 10:03:14 -0700
Engine • revision 5c984c26eb
Tools • Dart 2.18.2 • DevTools 2.15.0

B-Nagendra, 2022-10-12:
release version is better,still not good , glmark score is 114

HidenoriMatsubayashi, 2022-10-12:
> glmark score is 114

It's too low... Is GPU HW accelerator enabled?

B-Nagendra, 2022-10-12:
yes, will check why the score is low, thank you

HidenoriMatsubayashi, 2023-08-04:
For some reason, this issue seems to be improved, but I can see it on both Wayland and X11 now.

PR #363 (2023-08-07): egl: add async buffer swapping command option
Fixed #220
commit eb2687ad (2023-08-07): egl: add async buffer swapping command option (#363)

Fixed #220
closed 2023-08-07 by HidenoriMatsubayashi