sony/flutter-embedded-linux #226: [wayland] window decoration sometime doesn't appear in Weston 8

HidenoriMatsubayashi, 2021-09-22:
The window decoration sometimes doesn't appear in Weston 8 even if we use the `-d` option to show the decorations, but there are no problems with other Wayland compositors. I'm not sure the cause, but https://github.com/sony/flutter-embedded-linux/pull/221 change is affecting it.

### Related issues
- https://github.com/sony/flutter-embedded-linux/issues/6

psstoyanov, 2022-01-13:
Weird, with Weston 9 sometimes the decorations aren't displayed until it detects a mouse movement (haven't tried on touch enabled device with Weston for now). 
Once the movement is detected, the decoration buttons show up.
There is slight variance between the decoration strip showing and not but in both cases they appear once there is mouse movement.

Is this the same observation?
https://user-images.githubusercontent.com/3979134/149368076-0cf6f193-0944-46ee-99e3-a8da9a17d2d8.mp4

HidenoriMatsubayashi, 2022-01-14:
Thank you for your comment. Yes, it is.

HidenoriMatsubayashi, 2023-08-02:
Closing as it seems to work fine on top of master branch.

closed 2023-08-02 by HidenoriMatsubayashi