sony/flutter-embedded-linux #227: Building Galerry inside a ARM64 debian container

munoz0raul, 2021-10-04:
Hello, 

I'm trying to create a debian + wayland + flutter container.

I can run the sample app and also the [testing app](https://github.com/flutter/samples/tree/master/testing_app).

I don't know if I'm configuring everything in the right way, I'm still learning and trying to read all docs.

My issue is, trying to build the gallery app, I getting this error:

```
$ git clone https://github.com/flutter/gallery
$ cd gallery/
$ flutter-elinux config --enable-linux-desktop
Setting "enable-linux-desktop" value to "true".

You may need to restart any open editors for them to read new settings.
$ flutter-elinux build linux --debug
Running "flutter pub get" in gallery...                             8.2s

Building without sound null safety
For more information see https://dart.dev/null-safety/unsound-null-safety

ERROR: lib/studies/crane/item_cards.dart:47:21: Error: No named parameter with the name 'semanticsLabel'.
ERROR:                     semanticsLabel: destination.subtitleSemantics(context),
ERROR:                     ^^^^^^^^^^^^^^
ERROR: /opt/flutter-elinux/flutter/packages/flutter/lib/src/material/selectable_text.dart:176:9: Context: Found this candidate, but the arguments don't match.
ERROR:   const SelectableText(
ERROR:         ^^^^^^^^^^^^^^
ERROR: lib/studies/crane/item_cards.dart:71:19: Error: No named parameter with the name 'semanticsLabel'.
ERROR:                   semanticsLabel: destination.subtitleSemantics(context),
ERROR:                   ^^^^^^^^^^^^^^
ERROR: /opt/flutter-elinux/flutter/packages/flutter/lib/src/material/selectable_text.dart:176:9: Context: Found this candidate, but the arguments don't match.
ERROR:   const SelectableText(
ERROR:         ^^^^^^^^^^^^^^
Building Linux application...                                           
Build process failed
```

Any idea what could be wrong?

Thanks.

HidenoriMatsubayashi, 2021-10-05:
I got the same error. Also I got it even if I use the official flutter tool without flutter-linux. I guess this is an issue of between flutter/gallery version and flutter version.

galerry version: 426287b33df1b45168737779a6d0f23e73666be2

flutter vesion:
```
Flutter 2.5.2 • channel stable • https://github.com/flutter/flutter.git
Framework • revision 3595343e20 (4 days ago) • 2021-09-30 12:58:18 -0700
Engine • revision 6ac856380f
Tools • Dart 2.14.3
```

HidenoriMatsubayashi, 2021-10-05:
I'll close this issues because it is not an issue for the flutter embedder for the elinux and is not also my above comment. Thanks a lot.

closed 2021-10-05 by HidenoriMatsubayashi

munoz0raul, 2021-10-05:
Thanks @HidenoriMatsubayashi.

Do you know remember what version of the gallery used to work? Have seen other folks trying as well.

Also, do you have any other sample app nice for a demostration? I have a 7-inch display with iMX8 and I would like to have a demo to show the flutter capability for embedded applications. 

Thanks.

HidenoriMatsubayashi, 2021-10-05:
> Do you know remember what version of the gallery used to work? Have seen other folks trying as well.

I'm sorry I don't remember, but definitely it worked fine before. 

> Also, do you have any other sample app nice for a demostration? 

Let's say, [netflix-clone app](https://github.com/sopheamen007/app.mobile.netflix-clone-app-ui). You need to add the elinux-template (runner) and [the video player plugin for elinux](https://github.com/sony/flutter-elinux-plugins/tree/main/packages/video_player) into it, and you might need to modify some code but it worked fine on i.mx 8m platforms.

munoz0raul, 2021-10-05:
Hi @HidenoriMatsubayashi I will try the netflix app. Sorry didn´t find anything about "elinux-template (runner)" where is that?

Also, I'm trying some old versions and it is building but with a different problem:

```
root@imx8mm-var-som-symphony:~/gallery# git checkout e6a83f8e84b7457a18c3258ff420fcf64bb20cbd
flutter-elinux config --enable-linux-desktop
Setting "enable-linux-desktop" value to "true".

You may need to restart any open editors for them to read new settings.
root@imx8mm-var-som-symphony:~/gallery# cd flutter-embedded-linux/build/
bash: cd: flutter-embedded-linux/build/: No such file or directory
root@imx8mm-var-som-symphony:~/gallery# flutter-elinux build linux --debug
Running "flutter pub get" in gallery...                            31.9s

Building without sound null safety
For more information see https://dart.dev/null-safety/unsound-null-safety

Building Linux application... 
```                                          

```
root@imx8mm-var-som-symphony:~/flutter-embedded-linux/build# LD_LIBRARY_PATH=/opt/flutter-elinux/flutter/bin/cache/artifacts/engine/elinux-arm64-debug/ ./flutter-client -b /root/gallery/build/linux/arm64/debug/bundle -f
flutter: Observatory listening on http://127.0.0.1:34063/nCwrZJwdBd0=/
flutter: ════════
flutter: Warning: This application's locale, C_UTF-8, is not supported by all of its
flutter: localization delegates.
flutter: > A GalleryLocalizations delegate that supports the C_UTF-8 locale was not found.
flutter: > A MaterialLocalizations delegate that supports the C_UTF-8 locale was not found.
flutter: See https://flutter.dev/tutorials/internationalization/ for more
flutter: information about configuring an app's locale, supportedLocales,
flutter: and localizationsDelegates parameters.
flutter: ════════
flutter: 
flutter: Locale C_UTF-8 is not an available locale. Falling back to 'en'. Specify a different fallback locale with [LocaleNamesLocalizationsDelegate.fallbackLocale].
```

And the display is like that:
[image]

HidenoriMatsubayashi, 2021-10-05:
Please check [here](https://github.com/sony/flutter-elinux/wiki/Creating-flutter-projects#2-use-existing-official-flutter-projects)

munoz0raul, 2021-10-05:
Just new update about [gallery](https://github.com/flutter/gallery).

It is not working in the latest commit: `426287b33df1b45168737779a6d0f23e73666be2`

It is working well with the previous hash: `312e0cbd271c5da57256f874ae45f87d15103890`

Also, it just worked for me adding  `LC_ALL=en_US.UTF-8` in the running command:

`LD_LIBRARY_PATH=/opt/flutter-elinux/flutter/bin/cache/artifacts/engine/elinux-arm64-debug/ ./flutter-client -b /root/gallery/build/linux/arm64/debug/bundle -f`

munoz0raul, 2021-10-08:
Think as the container being a distribution. 

I made the debian distro with mesa work, now I'm trying to do the same with vivante libs.