sony/flutter-embedded-linux #236: How much memory does the flutter to run smoothly?

nanfang2000, 2021-11-16:
I have a low end SBC that have 512MB DDR and the GPU is mali 400. Can this support flutter?

psstoyanov, 2021-11-16:
It should be able to handle it. It runs on the PinePhone / A64LTS which also sport Mali 400 GPU so you don't have concerns there.
I don't have SBC with such restricted RAM - keep an eye on the usage there and adjust accordingly.

What's the SBC in question?

HidenoriMatsubayashi, 2021-11-16:
@psstoyanov
Thanks.

@nanfang2000 
Unfortunately, I don't know about SBC, too. But, Flutter requires OpenGL ES library. Also, it requires at least around 50 MB of memory (depends on your application).

nanfang2000, 2021-11-19:
@psstoyanov @HidenoriMatsubayashi 
thank you very much!
it's a cheap sbc use allwinner A33 CPU with 512MB ddr

I assume when in debug mode, 50MB isn't 3enough, right?

psstoyanov, 2021-11-19:
The dart server will take a lot more - I wouldn't program directly on the device if I were you given that it is using A7 cores on 40nm.
Compiling the project the first time will take a long while for sure on that device.

You can setup `zram` or `zswap` to combat the limited RAM available if you want to try to use the SBC by itself - some info from Arch's Wiki but apply to your OS of choice: https://wiki.archlinux.org/title/Improving_performance#zram_or_zswap

Running just the resulting bundles with `wayland` backend with the default project (haven't done anything to make it smaller or optimise it):
- `debug` -> `207MB`
- `release` -> `83MB`
Note: That's reported by `btop` so it can be inaccurate. Using DRM (`GBM` mode) directly could be a good idea if you aim for kiosk style of app and you don't need the rest the system dependencies.

When you take into account what else is running in `debug` mode (dart server/ observers/ etc), I think it will go outside of the available 512MB. Then on top you have some other necessities for the system itself - networking, audio and so on.

For release - you shouldn't have any issues.

During development, I would strongly recommend for you take a good look at remote execution on the target SBC given how weak it is:
https://github.com/sony/flutter-elinux/wiki/Remote-target-devices#4-run-your-flutter-apps-on-your-target-device
Remote execution should alleviate pretty much all of your problems as the SBC will be used just to run the resulting app with the heavy lifting being done elsewhere.
If you are pressed for memory - I would try with zram.

closed 2022-04-12 by HidenoriMatsubayashi