sony/flutter-embedded-linux #237: Mouse cursor is visible even when no mouse is connected

Taha-Firoz, 2021-11-24:
There is a mouse cursor that appears even when no mouse is connected, on flutter-pi it only comes up when a mouse is connected however in this case it comes up by default on. This is on a gbm backend

HidenoriMatsubayashi, 2021-11-25:
I've tested on Raspberry-Pi 4, but it works fine. The mouse cursor appears if a mouse is connected, and it doesn't appear if a mouse is not connected.

What device do you use? Please check your device environment settings, too.

Taha-Firoz, 2021-11-25:
I'm using a raspberry pi 4, and from what I remember the mouse wasn't there before. what can make the mouse cursor appear? I had recently run it on a freshly installed version of Ubuntu and the mouse was still appearing there. I'm using the remote install method, is there anyway I can investigate this further.

HidenoriMatsubayashi, 2021-11-26:
I'm using Ubuntu 21.04 version on raspberry pi 4. For drm backends, we use `udev` to detecte device hotplugs and inputs. See also: https://github.com/sony/flutter-embedded-linux/blob/be0a03f403cd78cf8299797818e56c4a1b362259/src/flutter/shell/platform/linux_embedded/window/elinux_window_drm.h#L331

Taha-Firoz, 2021-11-26:
Is there any way I can check some logs to confirm whether my ubuntu's generating ghost mouse events, which I'm probably sure it isn't.
This is how I'm building
```
flutter-elinux build elinux \
-v \
--target-backend-type gbm \
--target-arch=arm64 \
--system-include-directories=/usr/aarch64-linux-gnu/include/c++/9/aarch64-linux-gnu \
--target-sysroot=/{path}/ubuntu18-arm64-sysroot
```

This is how I'm installing
```
flutter-elinux run -d uperfect --release
```
Theres a bug in the run method it doesn't launch the actual ui, it crashes on getter called on null value. So I use the script only until install and then I exit the script when it asks for a password to launch,

```
ssh pi@pi
FLUTTER_DRM_DEVICE="/dev/dri/card1" ~/bin/{ui}/ui
```
This is how I run the project from my laptop, I SSH into the pi and running this command and the ui launches on the pi however it has the cursor in the default position, top left corner.

Taha-Firoz, 2021-11-26:
This is what happens, when I touch the screen the cursor appears and doesn't leave.

https://user-images.githubusercontent.com/42237761/143531250-23d9069e-1b31-4e30-84d2-ae0f9763a0a8.mp4

Taha-Firoz, 2021-11-26:
Okay another update I plugged in a mouse and when I unplugged the mouse the cursor leaves and doesnn't show up after I touch the screen. What do you think could be causing this ghost mouse movement, I never had this issue before on flutter-pi

Taha-Firoz, 2021-11-26:
Here is the output from `libinput debug-events`

I had just started the flutter ui and opened `libinput`
```
-event0   DEVICE_ADDED     TSTP MTouch                       seat0 default group1  cap:t  size 213x133mm ntouches 10 calib
-event1   DEVICE_ADDED     TSTP MTouch                       seat0 default group1  cap:k
-event2   DEVICE_ADDED     vc4                               seat0 default group2  cap:kp left scroll-nat scroll-button
-event3   DEVICE_ADDED     vc4                               seat0 default group2  cap:kp left scroll-nat scroll-button
```

This is the first touch event that shows the mouse pointer
```
-event0   TOUCH_DOWN       +0.000s      0 (0) 23.50/68.16 (50.17/91.00mm)
 event0   TOUCH_FRAME      +0.000s
 event0   TOUCH_UP         +0.100s
 event0   TOUCH_FRAME      +0.100s
```

This is when I plugged in the mouse and removed it
```
-event4   DEVICE_ADDED     Logitech B330/M330/M3             seat0 default group3  cap:kp left scroll-nat scroll-button
 event4   DEVICE_REMOVED   Logitech B330/M330/M3             seat0 default group3  cap:kp
```

These are subsequent touches that doesn't make the pointer show up
```
-event0   TOUCH_DOWN       +23.399s     0 (0)  4.76/ 2.00 (10.17/ 2.67mm)
 event0   TOUCH_FRAME      +23.399s
 event0   TOUCH_UP         +23.483s
 event0   TOUCH_FRAME      +23.483s
 event0   TOUCH_DOWN       +25.543s     0 (0)  6.48/ 7.49 (13.83/10.00mm)
 event0   TOUCH_FRAME      +25.543s
 event0   TOUCH_UP         +25.627s
 event0   TOUCH_FRAME      +25.627s
 event0   TOUCH_DOWN       +27.575s     0 (0) 27.71/72.78 (59.17/97.17mm)
 event0   TOUCH_FRAME      +27.575s
 event0   TOUCH_UP         +27.723s
 event0   TOUCH_FRAME      +27.723s
```

HidenoriMatsubayashi, 2021-11-26:
Thank you for your information. I'll do debugging next week and fix it.

HidenoriMatsubayashi, 2021-11-29:
@Taha-Firoz 

I found the cause of the bug, but I have no idea how to fix it now. I want to control whether the mouse cursor is displayed depending on whether the input device is a mouse or a touch device. Is there a good way?

Taha-Firoz, 2021-11-29:
Well this problem didn't exist on flutter-pi maybe we can look into how it's been implemented there. Their implementation looks a little different from the elinux way though.

https://github.com/ardera/flutter-pi/blob/master/src/user_input.c

HidenoriMatsubayashi, 2021-11-29:
Yes, I guess there isn't a way how to know a device whether the input device is a mouse or a touch device using `libinput`. I'll Consider a workaround.

HidenoriMatsubayashi, 2021-11-29:
BTW, you can use no-mouse cursor option if you don't need mouse inputs. See also: https://github.com/sony/flutter-embedded-linux/wiki/How-to-run-Flutter-apps#command-options

Taha-Firoz, 2021-11-29:
There doesn't seem to be a way to do that with a DRM backend though

Taha-Firoz, 2021-11-29:
> Yes, I guess there isn't a way how to know a device whether the input device is a mouse or a touch device using `libinput`. I'll Consider a workaround.

What about this line in that user_input file

```lang=c
  while (libinput_next_event_type(input->libinput) != LIBINPUT_EVENT_NONE) {
        event = libinput_get_event(input->libinput);
        event_type = libinput_event_get_type(event);

        switch (event_type) {
            case LIBINPUT_EVENT_DEVICE_ADDED:
                ok = on_device_added(input, event, timestamp);
                if (ok != 0) {
                    goto fail_destroy_event;
                }
                break;
            case LIBINPUT_EVENT_DEVICE_REMOVED:
                ok = on_device_removed(input, event, timestamp);
                if (ok != 0) {
                    goto fail_destroy_event;
                }
                break;
            case LIBINPUT_EVENT_KEYBOARD_KEY:
                ok = on_key_event(input, event);
                if (ok != 0) {
                    goto fail_destroy_event;
                }
                break;
            case LIBINPUT_EVENT_POINTER_MOTION:
                ok = on_mouse_motion_event(input, event);
                if (ok != 0) {
                    goto fail_destroy_event;
                }
                break;
	        case LIBINPUT_EVENT_POINTER_MOTION_ABSOLUTE:
                ok = on_mouse_motion_absolute_event(input, event);
                if (ok != 0) {
                    goto fail_destroy_event;
                }
                break;
	        case LIBINPUT_EVENT_POINTER_BUTTON:
                ok = on_mouse_button_event(input, event);
                if (ok != 0) {
                    goto fail_destroy_event;
                }
                break;
	        case LIBINPUT_EVENT_POINTER_AXIS:
                ok = on_mouse_axis_event(input, event);
                if (ok != 0) {
                    goto fail_destroy_event;
                }
                break;
            case LIBINPUT_EVENT_TOUCH_DOWN:
                ok = on_touch_down(input, event);
                if (ok != 0) {
                    goto fail_destroy_event;
                }
                break;
	        case LIBINPUT_EVENT_TOUCH_UP:
                ok = on_touch_up(input, event);
                if (ok != 0) {
                    goto fail_destroy_event;
                }
                break;
	        case LIBINPUT_EVENT_TOUCH_MOTION:
                ok = on_touch_motion(input, event);
                if (ok != 0) {
                    goto fail_destroy_event;
                }
                break;
	        case LIBINPUT_EVENT_TOUCH_CANCEL:
                ok = on_touch_cancel(input, event);
                if (ok != 0) {
                    goto fail_destroy_event;
                }
                break;
            case LIBINPUT_EVENT_TOUCH_FRAME:
                ok = on_touch_frame(input, event);
                if (ok != 0) {
                    goto fail_destroy_event;
                }
                break;
            default:
                break;
        }
        
        libinput_event_destroy(event);
    }
```

Taha-Firoz, 2021-12-01:
> @Taha-Firoz
> 
> > -event0   DEVICE_ADDED     TSTP MTouch                       seat0 default group1  cap:t  size 213x133mm ntouches 10 calib
> > -event1   DEVICE_ADDED     TSTP MTouch                       seat0 default group1  cap:k
> 
> The mouse cursor is displayed at this timing, but I have no idea how to fix it now. I want to control whether the mouse cursor is displayed depending on whether the input device is a mouse or a touch device. Do you know a good way to know what the input source is using libinput?

Could you reference where this is being done in `flutter-elinux` so I can see or try

HidenoriMatsubayashi, 2021-12-02:
@Taha-Firoz Thanks a lot.

closed 2021-12-02 by HidenoriMatsubayashi

Taha-Firoz, 2021-12-02:
Thanks @HidenoriMatsubayashi , can you just tell me how to pull this new commit, the tutorial for installing [flutter-elinux](https://github.com/sony/flutter-embedded-linux/wiki/Installing-flutter-elinux) uses the flutter-elinux repository but this is the `flutter-embedded-linux` repository. Does stuff get eventually merged from `flutter-embedded-linux` to `flutter-elinux` when theres supposed to be a release

HidenoriMatsubayashi, 2021-12-02:
> Does stuff get eventually merged from flutter-embedded-linux to flutter-elinux when theres supposed to be a release

That's right. Until then, please use the one built locally.