sony/flutter-embedded-linux #240: onLongPress not working with touch but does work with mouse

chetrf, 2021-12-03:
onLongPress using GestureDetector does not work for me with touch but does work with mouse.  onTap works fine.  Everything else seems fine also.  I am running Ubuntu 21.10 on Raspberry Pi 4GB RAM.  

jen@jen:~$ flutter-elinux --version
Flutter 2.5.3 • channel unknown • unknown source
Framework • revision 18116933e7 (7 weeks ago) • 2021-10-15 10:46:35 -0700
Engine • revision d3ea636dc5
Tools • Dart 2.14.4

HidenoriMatsubayashi, 2021-12-06:
Which backend are you using? Wayland or DRM?

chetrf, 2021-12-16:
Waylan

psstoyanov, 2022-01-21:
@chetrf , could you post some information about the setup you are using? 
My immediate thoughts are to check the touchscreen and if it passes the event as expected (possibly [libinput-debug-gui](https://manpages.ubuntu.com/manpages/hirsute/man1/libinput-debug-gui.1.html)).

Just gave it a go with PinePhone running Manjaro ARM (Plasma) and there wasn't a problem using `onLongPress` with `GestureDector`.

I've used the following examples to try it out: 
- `GestureDector` example: https://github.com/flutter-devs/flutter_gesture_detector/blob/main/lib/ui/home.dart
- Through `ElevatedButton`: https://googleflutter.com/flutter-button-onlongpress/ (swap `RaisedButton` with `ElevetatedButton` in this example)
<details>
<summary> flutter-elinux version </summary>

```
Flutter 2.8.1 • channel unknown • unknown source
Framework • revision 77d935af4d (5 weeks ago) • 2021-12-16 08:37:33 -0800
Engine • revision 890a5fca2e
Tools • Dart 2.15.1
```

</details>

<details>
<summary> PinePhone OS details </summary>

```
██████████████████  ████████   pak0@manjaro-arm 
██████████████████  ████████   ---------------- 
██████████████████  ████████   OS: Manjaro ARM Linux aarch64 
██████████████████  ████████   Host: Pine64 PinePhone (1.2) 
████████            ████████   Kernel: 5.16.1-1-MANJARO-ARM 
████████  ████████  ████████   Uptime: 1 hour, 44 mins 
████████  ████████  ████████   Packages: 887 (pacman) 
████████  ████████  ████████   Shell: bash 5.1.16 
████████  ████████  ████████   Resolution: 720x1440 
████████  ████████  ████████   Terminal: /dev/ttyS0 
████████  ████████  ████████   CPU: (4) @ 1.152GHz 
████████  ████████  ████████   Memory: 649MiB / 2994MiB 
████████  ████████  ████████
████████  ████████  ████████                
```
</details>

<details> 
<summary> plasmashell version </summary>

```
plasmashell 5.23.5
```
</details>

HidenoriMatsubayashi, 2022-08-21:
I'll close the issue as no response.

closed 2022-08-21 by HidenoriMatsubayashi