sony/flutter-embedded-linux #241: Terrible performance when running as non-root/non-interactive-user

Taha-Firoz, 2021-12-07:
I've made a new user that is in the `input` and `video` groups and can thus run the `flutter-elinux` application (running in drm), however when I run it as this new user the startup time and performance of the application is terrible. Do you have any idea why this is happening, I've tried looking around with `ulimit` and the outputs are different from the interactive login user and root.

output for new user(non shell/service account)
```
$ ulimit -a
time(seconds)        unlimited
file(blocks)         unlimited
data(kbytes)         unlimited
stack(kbytes)        8192
coredump(blocks)     0
memory(kbytes)       unlimited
locked memory(kbytes) 478699
process              13586
nofiles              1024
vmemory(kbytes)      unlimited
locks                unlimited
rtprio               0
```

output for root
```
root@pi:# ulimit -a
real-time non-blocking time  (microseconds, -R) unlimited
core file size              (blocks, -c) 0
data seg size               (kbytes, -d) unlimited
scheduling priority                 (-e) 0
file size                   (blocks, -f) unlimited
pending signals                     (-i) 13586
max locked memory           (kbytes, -l) 478699
max memory size             (kbytes, -m) unlimited
open files                          (-n) 1024
pipe size                (512 bytes, -p) 8
POSIX message queues         (bytes, -q) 819200
real-time priority                  (-r) 0
stack size                  (kbytes, -s) 8192
cpu time                   (seconds, -t) unlimited
max user processes                  (-u) 13586
virtual memory              (kbytes, -v) unlimited
file locks                          (-x) unlimited
```

The performance is fine when running as normal logged in user or root but it's unusable when running as non interactive user/ service account user.

HidenoriMatsubayashi, 2021-12-08:
I'm afraid I have no idea at the moment, but I don't guess it isn't flutter's issue.

Taha-Firoz, 2021-12-08:
🤔 😔

There should be something that's going on, the UI even starts up slower. It's under a second in root/interactive user and it's like 5-6 seconds when running as service account user.

Taha-Firoz, 2021-12-08:
Okay so I was advised to add the user into the `render` group to fix the performance and it's worked!.

closed 2021-12-08 by Taha-Firoz