sony/flutter-embedded-linux #243: GBM-DRM backend on  Pi4’s vc4-kms-v3d-pi4 overlay `Failed to create view controller`

Taha-Firoz, 2021-12-16:
I've been testing on hdmi displays so far but I've need to test on a raspberry pi that's has a MIPI DSI display using Pi4’s vc4-kms-v3d-pi4 overlay. When an hdmi display is connect the binary works correctly, but when it runs on the MPI DSI display it gives a `Failed to create view controller` error. Here is a line from the logs.

```
openat(AT_FDCWD, "/dev/dri/card0", O_RDWR|O_CLOEXEC) = 11
ioctl(11, DRM_IOCTL_MODE_GETRESOURCES, 0xffffcb8712b8) = -1 EOPNOTSUPP (Operation not supported)
close(11)                               = 0
```

HidenoriMatsubayashi, 2021-12-17:
> ioctl(11, DRM_IOCTL_MODE_GETRESOURCES, 0xffffcb8712b8) = -1 EOPNOTSUPP (Operation not supported)

It seems like `vc4-kms-v3d-pi4` problem. Does it support KMS fully?

Taha-Firoz, 2021-12-17:
Yup it's full KMS

HidenoriMatsubayashi, 2021-12-17:
Really? 

> ioctl(11, DRM_IOCTL_MODE_GETRESOURCES, 0xffffcb8712b8) = -1 EOPNOTSUPP (Operation not supported)

**Operation not supported** ..

Taha-Firoz, 2021-12-17:
can you tell me what operation that is so I can debug

Taha-Firoz, 2021-12-17:
> Really?
> 
> > ioctl(11, DRM_IOCTL_MODE_GETRESOURCES, 0xffffcb8712b8) = -1 EOPNOTSUPP (Operation not supported)
> 
> **Operation not supported** ..

Also it's running gnome shell perfectly fine and a QT application with an eglstream backend is also running correctly on it

Taha-Firoz, 2022-03-18:
@HidenoriMatsubayashi is there someway I can get `ELINUX` logs to print without recompiling the engine with the logs flag? I have to use `strace` to get any info. I'm interested in trying to get DRM working on this device, right now i'm using the wayland client as a work around. I'm using the (cutiepi)[https://cutiepi.io/] tablet and it seems to run ubuntu and raspberry pi os fine,

closed 2022-03-20 by Taha-Firoz