sony/flutter-embedded-linux #244: On-screen keyboard doesn't respect width and height

snke, 2021-12-21:
When running `flutter-client` with `-k -w somewidth -h someheight`,
the on-screen keyboard does not respect the width and height passed.

HidenoriMatsubayashi, 2021-12-27:
Thank you for your issue. What kind of Wayland compositor do you use? Actually, there are some improvement points in the on-screen keyboard like #126.

snke, 2022-01-10:
Thank you for your reply!

I am using Weston at the moment. I'll read through the other issues

psstoyanov, 2022-01-21:
@snke , I haven't looked into it too much but I think that's just how Weston deals with it by default.
Not sure if this will be of help but there seems to be different modes for keyboard used by Weston: https://gitlab.freedesktop.org/wayland/weston/-/commit/5ff978a52bcdb98f85d0932acce3eca6286dab09