sony/flutter-embedded-linux #245: GIF image display, increasing memory, unable to free memory.

dazuo0312, 2021-12-27:
Flutter -elinux runtime environment: ARM64, Debian, NXP iMX8  
The interface runs to display GIF images, and the memory continues to increase, starting at 286M, 12 hours later, up to 900M.
`class MyHomePage extends StatefulWidget {
  const MyHomePage({Key? key, required this.title}) : super(key: key);

  final String title;

  @override
  State<MyHomePage> createState() => _MyHomePageState();
}

class _MyHomePageState extends State<MyHomePage> {
  @override
  Widget build(BuildContext context) {
    return Scaffold(
      appBar: AppBar(
        title: Text(widget.title),
      ),
      body: Container(
        color: Colors.white,
        child: GridView.count(
          crossAxisCount: 3,
          childAspectRatio: 1.0,
          children: gifAnimation(),
        ),
      )
    );
  }

  List<Widget> gifAnimation() {
    return <Widget>[
      Image.asset(R.assetsAnimation),
      Image.asset(R.assetsAnimation),
      Image.asset(R.assetsAnimation),
    ];
  }
}`

nbhhcty, 2021-12-30:
I have the same problem

suyulin, 2022-01-04:
> free
It's not a wayland framework problem, flutter's 🐞。You can use frame animation to solve this problem。

dazuo0312, 2022-01-04:
> > free
> > It's not a wayland framework problem, flutter's 🐞。You can use frame animation to solve this problem。
Can you provide a demo?
I use frame animation, but the problem still exists.
Animationcontroller is used.
Webp animation also has this problem.
Running in X11 mode, there is no problem, only in Wayland mode.

suyulin, 2022-01-05:
Webp animation on Android , the memory is ok ?

dazuo0312, 2022-01-05:
> Webp animation on Android , the memory is ok ?

yes

suyulin, 2022-01-05:
> > Webp animation on Android , the memory is ok ?
> 
> yes

Flutter version is 2.5.2 ?

dazuo0312, 2022-01-06:
> > > Webp animation on Android , the memory is ok ?
> > 
> > 
> > yes
> 
> Flutter version is 2.5.2 ?

Flutter version is :2.5.0  or 2.8.1.

renehansen, 2022-04-11:
We have the same problem, and in addition to the approaches already mentioned, we can confirm that Lottie and Flare cause memory leaks as well, which means that there is basically no way of doing animations at the moment.  

Flutter version: 2.10.3

dazuo0312, 2022-04-14:
@HidenoriMatsubayashi  I used ARM64, Debian, NXP iMX8.
There is this problem on the arm64 board.

renehansen, 2022-04-20:
Yes, I am also experiencing the leak on arm64. Btw, here is the [source code](https://github.com/renehansen/leak_poc) that contains a frame animation, as well as Lottie, and Flare animations

HidenoriMatsubayashi, 2022-04-22:
Can anyone use debugging tools like `Valgrind` to investigate the memory leak points?

oz2mia, 2022-04-28:
Using valgrind does not work, the program will not start correct.
I have tried using google/sanitizers that are build in the flutter engine, used LSAN and ASAN, but they only report heap used,
but not where the leak is.

HidenoriMatsubayashi, 2023-12-11:
I see. Thanks. Would it be possible for you to send the PR?

closed 2023-12-11 by HidenoriMatsubayashi