sony/flutter-embedded-linux #246: Error while initializing the Dart VM: Snapshot not compatible with the current VM configuration

diybl, 2022-01-16:
LD_LIBRARY_PATH=./flutter-embedded-linux/build/  ./flutter-embedded-linux/build/flutter-client -b ~/pj/sample/build/elinux/x64/release/bundle

when i run the flutter app

it shows the follow error

[FATAL:flutter/runtime/dart_vm_initializer.cc(22)] Error while initializing the Dart VM: Snapshot not compatible with the current VM configuration: the snapshot requires 'product no-code_comments no-dwarf_stack_traces_mode lazy_async_stacks no-lazy_dispatchers use_bare_instructions dedup_instructions no-asserts x64-sysv compressed-pointers null-safety' but the VM has 'product no-code_comments no-dwarf_stack_traces_mode lazy_async_stacks no-lazy_dispatchers use_bare_instructions dedup_instructions no-asserts x64-sysv no-compressed-pointers null-safety'
Aborted (core dumped)

HidenoriMatsubayashi, 2022-01-18:
Thank you for your report.

This is due to `gen_snapshot`. I guess you use the pre-build data in https://github.com/sony/flutter-embedded-linux/releases/tag/40a99c5951. If so, please use `gen_snapshot` that you built yourself.

See also: https://github.com/sony/flutter-elinux/issues/70

Taha-Firoz, 2022-01-19:
@HidenoriMatsubayashi can you explain the work around? Debug mode has terrible performance and testing with it for now is very difficult.

HidenoriMatsubayashi, 2022-01-21:
> can you explain the work around?

Can you replace `${INSTALL_PATH}/flutter-elinux/flutter/bin/cache/artifacts/engine/android-arm64-*/linux-x64/gen_snapshot` to your `gen_snapshot ` that you built?
See also: https://github.com/sony/flutter-elinux/issues/75

psstoyanov, 2022-01-21:
Is it strange that I can observe the issue on my x64 desktop but not when compiling my arm devices  :thinking:

HidenoriMatsubayashi, 2022-01-24:
> Is it strange that I can observe the issue on my x64 desktop but not when compiling on my arm devices

Yes, that's right. This problem will happen when you build on x64 hosts. When building on an arm64 host, `gen_snapshot` for Linux is used. See: https://github.com/sony/flutter-elinux/blob/main/lib/commands/build.dart#L70

Taha-Firoz, 2022-01-26:
> > can you explain the work around?
> 
> Can you replace `${INSTALL_PATH}/flutter-elinux/flutter/bin/cache/artifacts/engine/android-arm64-*/linux-x64/gen_snapshot` to your `gen_snapshot ` that you built? See also: [sony/flutter-elinux#75](https://github.com/sony/flutter-elinux/issues/75)

I don't fully understand how to do this because I'm not really configuring a lot of stuff. All I do is provide an arm64 sysroot and I run build, how can I do this?

closed 2022-02-21 by HidenoriMatsubayashi