sony/flutter-embedded-linux #248: Question adout the example flutter-drm-gbm-backend  Support the rotation

andy2332, 2022-01-23:
Hello, do you have the intention of using DRM to support screen rotation?  Of course you can use DRM hardware to support rotation.  But it is expensive.  
I found in "flutter_embedder.h" : "
/// The transformation to apply to The render target before any rendering  
/// operations. This callback is optional.  
/// @attention When using a custom compositor, the layer offset and sizes  
/// will be affected by this transformation. It will be  
/// embedder responsibility to render contents at the  
/// transformed offset and size. This is useful for embedders  
/// that want to render transformed contents directly into  
/// hardware overlay planes without having to apply extra  
/// transformations to layer contents (which may necessitate an expensive off-screen   render pass).  
TransformationCallback surface_transformation;"  
Configure whether this option can quickly achieve screen rotation. In which file is it appropriate to add code? Thanks!

HidenoriMatsubayashi, 2022-01-24:
> Configure whether this option can quickly achieve screen rotation. In which file is it appropriate to add code?

I think that it is for PlatformViews implementation. So I think it's inappropriate to use it.

andy2332, 2022-01-24:
I have modified the test in the xx file, and the modifications are as follows:

andy2332, 2022-01-24:
I have modified the test in the flutter file, and the modifications are as follows:
 flutter_elinux_engine.cc:
 config.open_gl.surface_transformation = 
     [](void* user_data) -> FlutterTransformation {
  auto host = static_cast<FlutterELinuxEngine*>(user_data);
   return host->view()->on_get_transformation();
  };
flutter_elinux_view.cc:
FlutterTransformation FlutterELinuxView::on_get_transformation() {
  return my_view_to_display_transform;
}
I tested it on raspberry pi and it works, hope it can help you. Looking forward to you adding this feature！

andy2332, 2022-02-18:
I've added rotation (including touch rotation), do I need to send a code for reference?

HidenoriMatsubayashi, 2022-02-18:
It's OK. I'm working on it.

andy2332, 2022-02-18:
I send to your email

closed 2022-02-21 by HidenoriMatsubayashi