sony/flutter-embedded-linux #256: DRM GBM Rotation option is broken

Taha-Firoz, 2022-04-07:
I have a portrait first display and when I use the rotate option the flutter application rotates 90 degrees but the flutter app doesn't stretch into the new height width it only rotates by 90.

app without any rotation
[image]

app with -r 90 option
[image]

Taha-Firoz, 2022-04-07:
Here are my logs with the `-r 90` option
```
[INFO][ConfigureDisplay(67)] resolution: 800x1280
error: XDG_RUNTIME_DIR not set in the environment.
[INFO][CreateRenderSurface(143)] Display output resolution: 800x1280
```
I'm pretty the [problem is here](https://github.com/sony/flutter-embedded-linux/blob/a07efc4412820c6aebd8fa47e1eb40716bda9841/src/flutter/shell/platform/linux_embedded/window/elinux_window_drm.h#L131) You seem to swap the height and width values accordingly when there's a rotation but you don't use those height width values anywhere else. Finally the surface gets initialized with the original height width values instead of the swapped ones. 

What I don't understand is how the view is rotated by 90 degree if the surface is still at the wrong resolution.

```
    if (current_rotation_ == 90 || current_rotation_ == 270) {
      std::swap(width, height);
    }
    ...
    // swapped height width not used
    render_surface_ = native_window_->CreateRenderSurface();
    if (!render_surface_->SetNativeWindow(native_window_.get())) {
      return false;
    }

    if (view_properties_.view_mode != FlutterDesktopViewMode::kFullscreen) {
      ELINUX_LOG(WARNING)
          << "Normal mode is not supported, use fullscreen mode.";
      view_properties_.view_mode = FlutterDesktopViewMode::kFullscreen;
    }
    // Still using original Height Width values
    view_properties_.width = native_window_->Width();
    view_properties_.height = native_window_->Height();
    ELINUX_LOG(INFO) << "Display output resolution: " << view_properties_.width
                     << "x" << view_properties_.height;
```

HidenoriMatsubayashi, 2022-04-08:
Thank you for reporting the bug!

closed 2022-04-08 by HidenoriMatsubayashi

HidenoriMatsubayashi, 2022-04-08:
You need to build `libflutter_elinux_gbm.so` and replace old one to it.

## build with gbm backend (debug mode)

```shell
$ cmake -DBUILD_ELINUX_SO=ON -DBACKEND_TYPE=DRM-GBM -DCMAKE_BUILD_TYPE=Release -DENABLE_ELINUX_EMBEDDER_LOG=ON -DFLUTTER_RELEASE=OFF ..
$ cmake --build .
```

## build with gbm backend (release mode)

```shell
$ cmake -DBUILD_ELINUX_SO=ON -DBACKEND_TYPE=DRM-GBM -DCMAKE_BUILD_TYPE=Release -DENABLE_ELINUX_EMBEDDER_LOG=OFF -DFLUTTER_RELEASE=ON ..
$ cmake --build .
```

## relase old `libflutter_elinux_gbm.so` to the laster one

```shell
$ cp libflutter_elinux_gbm.so <path_to_flutter-elinux>/flutter/bin/cache/artifacts/engine/elinux-*-*/
```

See also: https://github.com/sony/flutter-embedded-linux/wiki/Building-Embedded-Linux-embedding-for-Flutter

Taha-Firoz, 2022-04-08:
Thank you I've understood this from the wiki already but I was more interested in understanding how to do cross compile, I already have an arm64 sysroot but how do I link it for flutter embedded linux.

Taha-Firoz, 2022-04-08:
@HidenoriMatsubayashi  I tried cross compiling with this after of course updating the `<path_to_target_sysroot>` in the cross-toolchain-aarch64 template file.

```
$ cmake -DBUILD_ELINUX_SO=ON -DBACKEND_TYPE=DRM-GBM -DCMAKE_BUILD_TYPE=Release -DENABLE_ELINUX_EMBEDDER_LOG=ON -DFLUTTER_RELEASE=ON -DCMAKE_TOOLCHAIN_FILE=/flutter-embedded-linux/cmake/cross-toolchain-aarch64-template.cmake
```
I first got this error 
```
fatal error: 'bits/c++config.h' file not found
#include <bits/c++config.h>
         ^~~~~~~~~~~~~~~~~~
```

I remember needing to add the system include directory to fix this for `flutter-elinux` builds so I added this line above  the set target sys root in `cross-toolchain-aarch64-template.cmake` 

```
include_directories(SYSTEM "/usr/aarch64-linux-gnu/include/c++/9/aarch64-linux-gnu")
```
Any where so the bits error is gone but now I'm getting this error

```
[ 97%] Building CXX object CMakeFiles/flutter_elinux_gbm.dir/src/flutter/shell/platform/common/client_wrapper/plugin_registrar.cc.o
make[2]: *** No rule to make target 'libflutter_engine.so', needed by 'libflutter_elinux_gbm.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/flutter_elinux_gbm.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

Taha-Firoz, 2022-04-10:
Okay I got it to build, `libflutter_engine.so` was missing, I also did need to add the `include_directories` in there or else it wouldn't build and lastly I had to remove `libm.a` from the sysroot because it wasn't letting cmake link directly to the `libm.so` file, removing it fixed the problem.