sony/flutter-embedded-linux #261: Pressing "2" key on keyboard produces error bug

barribarrier, 2022-05-29:
> Moved https://github.com/sony/flutter-elinux/issues/92 to here. Think it belongs here.

Flutter version: 2.10.3

```bash
flutter-elinux create keyboard_test
cd keyboard_test
flutter-elinux run
```

Press "2" key (not numpad) on keyboard.
```
Launching lib/main.dart on eLinux in debug mode...
Building an eLinux application with wayland backend in debug mode for x64 target...        15.1s
Syncing files to device eLinux...                                  348ms

Flutter run key commands.
r Hot reload. 🔥🔥🔥
R Hot restart.
h List all available interactive commands.
d Detach (terminate "flutter run" but leave application running).
c Clear the screen
q Quit (terminate the application on the device).

💪 Running with sound null safety 💪

An Observatory debugger and profiler on eLinux is available at: http://127.0.0.1:40117/q_XKyqr8RUQ=/
The Flutter DevTools debugger and profiler on eLinux is available at:
http://127.0.0.1:9101?uri=http://127.0.0.1:40117/q_XKyqr8RUQ=/

══╡ EXCEPTION CAUGHT BY SERVICES LIBRARY ╞══════════════════════════════════════════════════════════
The following assertion was thrown during a platform message callback:
Attempted to send a key down event when no keys are in keysPressed. This state can occur if the key
event being sent doesn't properly set its modifier flags. This was the event:
RawKeyDownEvent#57305(logicalKey: LogicalKeyboardKey#00032(keyId: "0x00000032", keyLabel: "2",
debugName: "Digit 2"), physicalKey: PhysicalKeyboardKey#700e1(usbHidUsage: "0x000700e1", debugName:
"Shift Left"), repeat: false) and its data: RawKeyEventDataLinux#b18e7(toolkit: GLFW,
unicodeScalarValues: 50, scanCode: 50, keyCode: 50, modifiers: 0, isDown: true)
'package:flutter/src/services/raw_keyboard.dart':
Failed assertion: line 685 pos 7: 'event is! RawKeyDownEvent || _keysPressed.isNotEmpty'

Either the assertion indicates an error in the framework itself, or we should provide substantially
more information in this error message to help you determine and fix the underlying cause.
In either case, please report this assertion by filing a bug on GitHub:
  https://github.com/flutter/flutter/issues/new?template=2_bug.md

When the exception was thrown, this was the stack:
#2      RawKeyboard.handleRawKeyEvent (package:flutter/src/services/raw_keyboard.dart:685:7)
#3      KeyEventManager.handleRawKeyMessage (package:flutter/src/services/hardware_keyboard.dart:818:33)
#4      BasicMessageChannel.setMessageHandler.<anonymous closure> (package:flutter/src/services/platform_channel.dart:77:49)
#5      BasicMessageChannel.setMessageHandler.<anonymous closure> (package:flutter/src/services/platform_channel.dart:76:47)
#6      _DefaultBinaryMessenger.setMessageHandler.<anonymous closure> (package:flutter/src/services/binding.dart:377:35)
#7      _DefaultBinaryMessenger.setMessageHandler.<anonymous closure> (package:flutter/src/services/binding.dart:374:46)
#8      _invoke2.<anonymous closure> (dart:ui/hooks.dart:190:15)
#12     _invoke2 (dart:ui/hooks.dart:189:10)
#13     _ChannelCallbackRecord.invoke (dart:ui/channel_buffers.dart:42:5)
#14     _Channel.push (dart:ui/channel_buffers.dart:132:31)
#15     ChannelBuffers.push (dart:ui/channel_buffers.dart:329:17)
#16     PlatformDispatcher._dispatchPlatformMessage (dart:ui/platform_dispatcher.dart:583:22)
#17     _dispatchPlatformMessage (dart:ui/hooks.dart:89:31)
(elided 5 frames from class _AssertionError and dart:async)
```

No error is produced when running the same project with flutter linux.
```bash
flutter run -d linux
```

barribarrier, 2022-05-29:
Actually I found out that with a Tenkeyless keyboard, pressing "2" produces above error.
But with a full sized keyboard, pressing "2" does not produce error.

However, pressing "m" key on both tenkeyless and standard sized keyboard produces error.
Guaranteed way to reproduce this error is to press "m" key.

tronje, 2025-01-03:
I experienced the same bug, and found a solution. The following patch for flutter-embedded-linux fixes this:

```diff
diff --git a/src/flutter/shell/platform/linux_embedded/plugins/key_event_plugin.cc b/src/flutter/shell/platform/linux_embedded/plugins/key_event_plugin.cc
index 923054b..b1fcafb 100644
--- a/src/flutter/shell/platform/linux_embedded/plugins/key_event_plugin.cc
+++ b/src/flutter/shell/platform/linux_embedded/plugins/key_event_plugin.cc
@@ -132,7 +132,6 @@ void KeyeventPlugin::SendKeyEvent(uint32_t keycode,
   event.AddMember(kKeyCodeKey, keycode, allocator);
   event.AddMember(kKeyMapKey, kLinuxKeyMap, allocator);
   event.AddMember(kToolkitKey, kGLFWKey, allocator);
-  event.AddMember(kScanCodeKey, keycode, allocator);
   event.AddMember(kModifiersKey, modifiers, allocator);
   if (unicode != 0) {
     event.AddMember(kUnicodeScalarValues, unicode, allocator);
```

The reason this works is that , in Flutter, [here](https://github.com/flutter/flutter/blob/17025dd88227cd9532c33fa78f5250d548d87e9a/packages/flutter/lib/src/services/raw_keyboard_linux.dart#L94), the scan code is used to retrieve a physical key code, and then [here](https://github.com/flutter/flutter/blob/17025dd88227cd9532c33fa78f5250d548d87e9a/packages/flutter/lib/src/services/keyboard_maps.g.dart#L1995) the scan code of the "2" key leads to the `PhysicalKeyboardKey` for left-shift. This is a modifier key, and that messes with the tracking logic for key-up/key-down in the `hardware_keyboard` code, and trips the assertion that causes the exception you're seeing.

@HidenoriMatsubayashi you're not accepting pull requests, but please feel free to just take the above patch or a similar fix to address this issue, if you like. Thank you.

HidenoriMatsubayashi, 2025-01-03:
> the scan code is used to retrieve a physical key code, 

The patch may fix the issue, but for regular key input cases, does it work? `PhysicalKeyboardKey` needs/uses scan code, right?

> you're not accepting pull requests,

You can create and send a PR now :)