sony/flutter-embedded-linux #263: How to build flutter-embedded-linux with cmake

apy7hst, 2022-06-19:
Cmakelists.txt says that the minimum cmake version required is 3.10 but I have a higher version of cmake already installed but it is still not building on my linux. What is the solution for this?

HidenoriMatsubayashi, 2022-06-20:
Can you please share the error log?

apy7hst, 2022-06-22:
here is the error log :-

`CMake Error at CMakeLists.txt:10 (project):
  The CMAKE_CXX_COMPILER:`
  
   `clang++`

  `is not a full path and was not found in the PATH.`

  `Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.`

`CMake Error at CMakeLists.txt:10 (project):
  The CMAKE_C_COMPILER:`

    clang

  `is not a full path and was not found in the PATH.`

  `Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.`
`

I was building flutter for ubuntu touch OS because flutter does not exist for ubuntu touch OS using the command `clickable`. If you want to learn what is clickable, head to this link :- https://clickable-ut.dev/en/latest

HidenoriMatsubayashi, 2022-06-22:
Did you install clang?

HidenoriMatsubayashi, 2022-06-22:
Please check https://github.com/sony/flutter-embedded-linux/wiki/Building-Embedded-Linux-embedding-for-Flutter.

closed 2022-06-22 by HidenoriMatsubayashi

apy7hst, 2022-06-27:
while building, it threw one more error, here is the log :- 

`--   Requested 'wayland-client >= 1.16.0' but version of Wayland Client is 1.12.0
-- Configuring incomplete, errors occurred!`

Is there a fix for this? I've read the dependencies wiki already and it says libwayland-dev is enough but libwayland-dev is already installed in my system.

HidenoriMatsubayashi, 2022-06-27:
> Requested 'wayland-client >= 1.16.0' but version of Wayland Client is 1.12.0

I'm not sure your environment, but please install wayland-client verion 1.16.0 or above.