sony/flutter-embedded-linux #266: cmake gives linker error while performing cmake --build .

Khalid-Faisal, 2022-06-22:
I am following steps mentioned in wiki, I tried prebuilt libflutter_engine.so (elinux-arm64-debug, elinux-arm64-release, elinux-x64), but every time, cmake is throwing same issue, I am sharing error here.

```
ubuntu@ubuntu:build$ sudo cmake -DUSER_PROJECT_PATH=examples/flutter-wayland-client -DCMAKE_BUILD_TYPE=Debug ..
User project: examples/flutter-wayland-client
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/flutter-projects/mylxc_flutter/flutter-embedded-linux/build
ubuntu@ubuntu:build$ sudo cmake --build .
[  1%] Linking CXX executable flutter-client
/usr/bin/ld: libflutter_engine.so: error adding symbols: file in wrong format
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [CMakeFiles/flutter-client.dir/build.make:857: flutter-client] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/flutter-client.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

NOTE: I have installed Yocto for cross building [Cross Building from x64 to arm64](https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#3-cross-building-from-x64-to-arm64)

Thanks in advance

HidenoriMatsubayashi, 2022-06-22:
> /usr/bin/ld: libflutter_engine.so: error adding symbols: file in wrong format
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Are you building on x64 for x64? or on x64 for arm64?

closed 2022-06-22 by HidenoriMatsubayashi

HidenoriMatsubayashi, 2022-06-22:
If you want to self-build on x64 -> use elinux-x64 (e.g. https://github.com/sony/flutter-embedded-linux/releases/download/f15f824b57/elinux-x64-release.zip)
If you want to cross-build on x64 for arm64 -> see: https://github.com/sony/flutter-embedded-linux/wiki/Building-Embedded-Linux-embedding-for-Flutter#cross-build

Khalid-Faisal, 2022-06-22:
Sir,
Thank you so much for your valuable response,
Actually, I want to run flutter application on LE device, I am following wiki [flutter-embedded-linux-wiki](https://github.com/sony/flutter-embedded-linux/wiki), 

- I am able install [flutter-elinux](https://github.com/sony/flutter-embedded-linux/wiki/Installing-flutter-elinux) properly. 
- Earlier I thought of skipping building [Building Embedded Linux embedding for Flutter](https://github.com/sony/flutter-embedded-linux/wiki/Building-Embedded-Linux-embedding-for-Flutter). 
- I directly built an application by following [Building Flutter apps](https://github.com/sony/flutter-embedded-linux/wiki/Building-Flutter-apps), then I followed [Cross Building from x64 to arm64](https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#3-cross-building-from-x64-to-arm64), I used [Yocto SDK]([Use Yocto SDK](https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#case-2-use-yocto-sdk)) for cross building.  
- Later I realized that I need flutter-client to run application on LE device.
- Now I am following [Building Embedded Linux embedding for Flutter](https://github.com/sony/flutter-embedded-linux/wiki/Building-Embedded-Linux-embedding-for-Flutter). I tried what you suggested to do, I executed [cross-build on x64 for arm64](https://github.com/sony/flutter-embedded-linux/wiki/Building-Embedded-Linux-embedding-for-Flutter#cross-build) but got an issue. I am putting logs below,

```
ubuntu@ubuntu:cmake$ sudo cmake -DUSER_PROJECT_PATH=examples/flutter-wayland-client -DCMAKE_TOOLCHAIN_FILE=cmake/cross-toolchain-aarch64-template.cmake ..
Enter password for afaisal (QUALPASS):
-- The CXX compiler identification is Clang 10.0.0
-- The C compiler identification is Clang 10.0.0
-- Check for working CXX compiler: /usr/bin/clang++
-- Check for working CXX compiler: /usr/bin/clang++ -- broken
CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCXXCompiler.cmake:53 (message):
  The C++ compiler

    "/usr/bin/clang++"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /local/flutter-projects/flutter-embedded-linux/cmake/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/make cmTC_71a20/fast && /usr/bin/make -f CMakeFiles/cmTC_71a20.dir/build.make CMakeFiles/cmTC_71a20.dir/build
    make[1]: Entering directory '/local/flutter-projects/flutter-embedded-linux/cmake/CMakeFiles/CMakeTmp'
    Building CXX object CMakeFiles/cmTC_71a20.dir/testCXXCompiler.cxx.o
    /usr/bin/clang++ --target=aarch64-linux-gnu --sysroot=/usr/bin/    -std=gnu++17 -o CMakeFiles/cmTC_71a20.dir/testCXXCompiler.cxx.o -c /local/flutter-projects/flutter-embedded-linux/cmake/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
    Linking CXX executable cmTC_71a20
    /usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_71a20.dir/link.txt --verbose=1
    /usr/bin/clang++ --target=aarch64-linux-gnu --sysroot=/usr/bin/       CMakeFiles/cmTC_71a20.dir/testCXXCompiler.cxx.o  -o cmTC_71a20 
    /usr/bin/ld: unrecognised emulation mode: aarch64linux
    Supported emulations: elf_x86_64 elf32_x86_64 elf_i386 elf_iamcu elf_l1om elf_k1om i386pep i386pe
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    make[1]: *** [CMakeFiles/cmTC_71a20.dir/build.make:87: cmTC_71a20] Error 1
    make[1]: Leaving directory '/local/flutter-projects/flutter-embedded-linux/cmake/CMakeFiles/CMakeTmp'
    make: *** [Makefile:121: cmTC_71a20/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:10 (project)

-- Configuring incomplete, errors occurred!
See also "/local/flutter-projects/flutter-embedded-linux/cmake/CMakeFiles/CMakeOutput.log".
See also "/local/flutter-projects/flutter-embedded-linux/cmake/CMakeFiles/CMakeError.log".

```

HidenoriMatsubayashi, 2022-06-22:
> -DCMAKE_TOOLCHAIN_FILE=cmake/cross-toolchain-aarch64-template.cmake

Did you change it correctly and setup Yocto SDK?

Khalid-Faisal, 2022-06-22:
Yes I changed it as mentioned in [Cross Build](https://github.com/sony/flutter-embedded-linux/wiki/Building-Embedded-Linux-embedding-for-Flutter#cross-build). I followed [meta-flutter](https://github.com/sony/meta-flutter#cross-building-using-yocto-sdk) for Yocto setup

Khalid-Faisal, 2022-06-22:
> Also, I recommend you to use flutter-elinux to build your apps. See: https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#case-2-use-yocto-sdk

Yes, I followed [flutter-elinux ](https://github.com/sony/flutter-elinux/wiki/Building-flutter-apps#case-2-use-yocto-sdk) and I changed _**cross-toolchain-aarch64-template.cmake**_ file as mentioned in [Cross Build](https://github.com/sony/flutter-embedded-linux/wiki/Building-Embedded-Linux-embedding-for-Flutter#cross-build). I followed [meta-flutter](https://github.com/sony/meta-flutter#cross-building-using-yocto-sdk) for Yocto setup

HidenoriMatsubayashi, 2022-06-22:
You don't need to use "cross-toolchain-aarch64-template.cmake " when you use `flutter-elinux`. Can you use `flutter-elinux` only?

HidenoriMatsubayashi, 2022-06-22:
See https://github.com/sony/flutter-elinux/wiki/Remote-target-devices

Khalid-Faisal, 2022-06-24:
Thank you so much @HidenoriMatsubayashi, It helped me. App is working now!