sony/flutter-embedded-linux #272: Add support for building for arm64

apy7hst, 2022-07-24:
Please add support for building flutter for embedded linux for arm64 as well. This repository supports building only for amd64.

HidenoriMatsubayashi, 2022-07-24:
I'm not sure, but developers can cross-build...

closed 2022-07-24 by HidenoriMatsubayashi

apy7hst, 2022-07-25:
> I'm not sure, but developers can cross-build...

Even I am trying to develop this for ubuntu touch and it is arm64. But it fails when I build it using the arm64 libffi7 package, and it says the problem is it is amd64.

apy7hst, 2022-07-25:
> I guess it's a problem on your environment (os). Because I can build this software on arm64 devices such as Raspberry Pi4. Also, this software doesn't have any implementation of CPU arch dependencies.

The host os I am building this on is ubuntu touch