sony/flutter-embedded-linux #281: flutter-client fails to start Dart VM HTTP Service

LeoBound, 2022-08-11:
Hi,

I've built the debug wayland recipes using meta-flutter and can run flutter apps in debug mode.
Whenever I start a flutter app (by using `flutter-client -b /path/to/bundle`) I get the following error
```
flutter: Could not start Dart VM service HTTP server:
SocketException: Failed to create server socket (OS Error: Cannot assign requested address, errno = 99), address = 127.0.0.1, port = 0
#0      _NativeSocket.bind (dart:io-patch/socket_patch.dart:996:7)
<asynchronous suspension>
#1      Server.startup.startServer (dart:vmservice_io/vmservice_server.dart:424:19)
<asynchronous suspension>
#2      Server.startup (dart:vmservice_io/vmservice_server.dart:443:11)
<asynchronous suspension>
```

and subsequently I am unable to attach the debugger.
This also fails when using flutter custom-devices on a host machine to try and debug (same error).

Why would this be, is there a way to see a more verbose log?

Thanks

HidenoriMatsubayashi, 2022-08-12:
Are you using docker? If yes, does [this issue](https://github.com/dart-lang/sdk/issues/44787) help you?

LeoBound, 2022-08-12:
Unfortunately not - I'm running it on some embedded hardware where I added meta-flutter to the yocto build process.

HidenoriMatsubayashi, 2022-08-12:
Sorry, I can't debug any further as I don't know your environment. Are you using [flutter-elinux](https://github.com/sony/flutter-elinux)?

LeoBound, 2022-08-12:
The bundle was created with `flutter build bundle` with the standard flutter tool. 
After these errors the application does run as expected - there is just no way to debug it.
Could you let me know what details you need to know about the environment? It's a custom board with an iMX6.

Thanks

HidenoriMatsubayashi, 2022-08-12:
Are you trying to attach your flutter app from your host PC or just launching it on arm64 devices?

HidenoriMatsubayashi, 2022-08-12:
> After these errors the application does run as expected - there is just no way to debug it.

I think that the settings related to the network/port of the device you are using are affecting the error.

LeoBound, 2022-08-16:
Do you know where could I find more verbose logs?  

The logs in my first post are the output of `flutter-client -f -b /path/to/bundle`. Does flutter-client or the flutter engine output any other logs, I couldn't find any in /var/logs.

My steps were:
- Add meta-flutter to my yocto distro and building in debug mode
- Upload the image to the device
- Add the device with flutter custom-devices with the flutter SDK (3.0.5)
  - for the command I launch it with `XDG_RUNTIME_DIR=/tmp/-runtime-dir flutter-client -f -b /tmp/${appName}`
- I get the error in the log about being unable to create the server socket.

I'm using dropbear for SSH access instead of OpenSSH, could this be why?

HidenoriMatsubayashi, 2022-08-16:
> Add the device with flutter custom-devices with the flutter SDK (3.0.5) 

Why are you using the official flutter SDK? Could you try using [flutter-elinux](https://github.com/sony/flutter-elinux/wiki/Remote-target-devices) once?

HidenoriMatsubayashi, 2023-08-04:
Closing as no update for a while.

closed 2023-08-04 by HidenoriMatsubayashi