sony/flutter-embedded-linux #282: Add JSON config/ini file to launch instead of command line only

Taha-Firoz, 2022-08-13:
I've worked in quiet a few designs in possible config files to launch the embedder from instead of the just command line parameters. It could help make it a lot clearer to start your application and configurations can be more detailed if we do.

HidenoriMatsubayashi, 2022-08-20:
I think YAML is better than JSON for the config file.

Taha-Firoz, 2022-08-20:
True YAML is easier to write but I was pitching JSON because you already have RapidJson in your project which we could use to parse the file with.

HidenoriMatsubayashi, 2022-08-20:
RapidJson is too much to parse config files. And also we need to include it to the template of flutter-elinux if we use it.

Taha-Firoz, 2022-08-20:
So we just do it through flutter-elinux since it has json parsing included in it, or can we pull in a yaml parsing dep?

HidenoriMatsubayashi, 2022-08-20:
Technically, we have two options. First one is to parse the config file on flutter-elinux side and reflect it on the command line　options. Second one is to implement a simple parser with embedder or template source code (Not flutter-elinux side).

Taha-Firoz, 2022-08-20:
I actually want to add something more with the config file, basically I want to add more control over DKMS. I'm having a similar conversation over in [the flutter-pi embedder](https://github.com/ardera/flutter-pi/issues/201) right now. It would be slightly easier for me to work on this because I am more familiar with the flutter-embedded-linux because I've basically been working with this project over 5-6 months at this point.

HidenoriMatsubayashi, 2022-08-20:
> basically I want to add more control over DKMS

Please create another issue.