sony/flutter-embedded-linux #288: Flutter 3.3.0 artifacts release

HidenoriMatsubayashi, 2022-09-01:
Hi @taka-okayama, 

Official flutter 3.3.0 was released two days ago (https://docs.flutter.dev/development/tools/sdk/releases?tab=linux), but the artifacts for elinux hasn't released yet (https://github.com/sony/flutter-embedded-linux/releases). Can you please take a look the internal infra/CI?

taka-okayama, 2022-09-01:
@HidenoriMatsubayashi 

I tried to compile flutter engine **in local environment**, but failed. We use these options to build flutter engine in our CI.

- `flutter/tools/gn --target-os linux --linux-cpu x64 ...` -> Compile Error
- `flutter/tools/gn --target-os linux --linux-cpu arm64 ...` -> Success
- `flutter/tools/gn ...`(no target option) -> Success

Could you try these options in your environment? I'm not sure whether this problem depends on my environment or not.

taka-okayama, 2022-09-01:
Error Log

```
Defining target //third_party/angle/samples:post_sub_buffer(//build/toolchain/linux:clang_x64)
Running //third_party/angle/src/tests/BUILD.gn with toolchain //build/toolchain/linux:clang_x64
Defining target //third_party/angle/samples:simple_instancing(//build/toolchain/linux:clang_x64)
Loading //build/config/chromecast_build.gni (referenced from //third_party/angle/src/tests/BUILD.gn:5)
ERROR at //third_party/angle/src/tests/BUILD.gn:5:1: Can't load input file.
import("//build/config/chromecast_build.gni")
^-------------------------------------------
Unable to load:
  /home/vagrant/engine/src/build/config/chromecast_build.gni
I also checked in the secondary tree for:
  /home/vagrant/engine/src/build/secondary/build/config/chromecast_build.gni
See //third_party/angle/BUILD.gn:74:7: which caused the file to be included.
      "$angle_root/src/tests:angle_tests",
      ^----------------------------------
```

HidenoriMatsubayashi, 2022-09-02:
@taka-okayama thanks.

> flutter/tools/gn --target-os linux --linux-cpu x64 ... -> Compile Error

Can you check https://github.com/sony/flutter-embedded-linux/wiki/Building-Flutter-Engine-from-source#4-build-embedder and try it again?

[image]

taka-okayama, 2022-09-02:
@HidenoriMatsubayashi 

thank you for your information.
I fixed CI code to remove `--target-os` option and run CI.

3.3.0 binary has been uploaded. Can you confirm it?

HidenoriMatsubayashi, 2022-09-02:
@taka-okayama  Looks good. Thanks a lot. Closing.

closed 2022-09-02 by HidenoriMatsubayashi