sony/flutter-embedded-linux #289: Building flutter-engine with fontconfig support

andreadaoud, 2022-09-08:
If we want to display CJK characters using system font, we have to enable fontconfig support. Just add `--enable-fontconfig` to `./flutter/tools/gn` command. Could you please consider enabling this in your CI pipeline when you build the artifacts?

HidenoriMatsubayashi, 2022-09-10:
Hi @taka-okayama 

This request is understandable/reasonable to me. 

> Just add --enable-fontconfig to ./flutter/tools/gn command. Could you please consider enabling this in your CI pipeline when you build the artifacts?

Can you please address that? All you have to do is add `--enable-fontconfig` to the build script in the CI.

Updated to add --enable-fontconfig: 
https://github.com/sony/flutter-embedded-linux/wiki/Building-Flutter-Engine-from-source#4-build-embedder

HidenoriMatsubayashi, 2022-10-06:
@taka-okayama ping

taka-okayama, 2022-10-06:
@HidenoriMatsubayashi
sorry, we have a problem in CI now.

HidenoriMatsubayashi, 2022-10-06:
@taka-okayama Ok, then please change the recipe after the CI is green.

taka-okayama, 2022-10-07:
https://github.com/flutter/flutter/issues/112919

HidenoriMatsubayashi, 2022-10-07:
Thanks. Commented.

taka-okayama, 2022-10-13:
@andreadaoud @HidenoriMatsubayashi 

I've updated artifacts with `--enable-fontconfig`. Could you confirm?

HidenoriMatsubayashi, 2022-10-13:
@taka-okayama 

You mean you updated https://github.com/sony/flutter-embedded-linux/releases/tag/c08d7d5efc and its build includes that option? Or you just changed only internal recipe change?

taka-okayama, 2022-10-13:
@HidenoriMatsubayashi 

oh, sorry. Local CI completed but deployment step in CI was skipped because of no change in flutter version. I'm changing CI script to re-build artifacts forcibly(option).

HidenoriMatsubayashi, 2022-10-13:
Understood, but it's OK for me to wait until next release.

andreadaoud, 2022-10-13:
I've already compiled myself, so waiting until next release is also okay for me. But I'm willing to test anytime.

taka-okayama, 2022-10-14:
@HidenoriMatsubayashi @andreadaoud 

thanks for your patience. this change will be deployed next release.

HidenoriMatsubayashi, 2022-10-21:
@taka-okayama Confirmed that this option was included in the artifacts for v3.3.5. Thanks. I'll close the issue.

closed 2022-10-21 by HidenoriMatsubayashi